microsoft/AirSim #2005: Get Actors information from server side ( bbox for animals in Camera trap )

eslambakr, 2019-06-06:
Hello,

Thanks for this great effort.
I want to use CameraTrap Env. to collect data for animals but I can't get any information except the images it self, I can't get any information about the animals in the scene. 
For example Segmentation for the animals or bounding boxes for them.

I will appreciate if any one could help.

Thanks in advance.

msb336, 2019-06-06:
It looks like `simListSceneObjects`, `simGetObjectPose`, and `simSetSegmentationID` will be helpful functions for you.

`simListSceneObjects()` will return a list of ever object in the scene. From there, you can pick whatever animal you desire from the list, and give it a segmentation ID using `simSetSegmentationID`. `simGetObjectPose` will return your animals location and orientation in the scene.

eslambakr, 2019-06-06:
Thanks a lot for your quick replay and your help

when i am using the following line 
objects = client.simListSceneObjects('.*?DeerBoth.*?')

Unfortunately, I got this error 
RPCError: rpclib: server could not find function 'simListSceneObjects' with argument count 1.

By the way i set in my setting this parameters as follows (I list them as I think the error in them):
"SimMode": "ComputerVision",
  "RpcEnabled": true,

msb336, 2019-06-06:
Looks like that environment is a little outdated. I'll look into updating it. I'll post in this thread once it has been updated with the new apis

eslambakr, 2019-06-06:
OK, Thanks a lot I will wait your update .

msb336, 2019-06-07:
@eslambakr just uploaded Trapcam to v1.2.2 windows releases. Let me know if that solves issue!

closed 2019-06-10 by msb336