microsoft/AirSim #2006: Fatal error when collecting images

sophiefsadler, 2019-06-07:
I get the following error when collecting images from Airsim:

[image]

The code I am running is the following. Note that no error is raised in Python when this occurs; my code just hangs after the Unreal window has closed after crashing.

```
"""
Usage:
    data_generate.py [options] POSITIONS_DIR DATA_DIR

Options:
    -h --help                       Show this screen
    --n_positions=<n>               Number of spawn positions. At each position 1 image is
                                    generated unless the car collides [default: 300]

Arguments:
    POSITIONS_DIR   Path to a directory containing at least one yaml file containing a list of
                    AirSim Poses
    DATA_DIR        Name of the folder where the generated images and the yaml file containing
                    information about the images should be saved
"""
import os
import io
import random
import time
import logging

from docopt import docopt
import yaml
import numpy as np
from PIL import Image

import setup_path 
import airsim

def retrieve_coordinates_list(positions_dir):
    positions_yaml_files = [yaml_file for yaml_file in os.listdir(positions_dir) if yaml_file.endswith('.yml')]
    airsim_poses = []
    for yaml_file in positions_yaml_files:
        yaml_path = positions_dir + '/' + yaml_file
        positions_list = yaml.load(stream=open(yaml_path, "r"), Loader=yaml.FullLoader)
        airsim_poses = airsim_poses + positions_list
    return airsim_poses

def random_spawn(airsim_poses, client):
    num_positions = len(airsim_poses)
    position_index = random.randint(0, num_positions - 1)
    chosen_pose = airsim_poses[position_index]
    client.simSetVehiclePose(chosen_pose, True)
    weather_time_dict = random_weather_and_time(client)
    collision_info = client.simGetCollisionInfo()
    return chosen_pose, collision_info, weather_time_dict

def random_weather_and_time(client):
    possible_times = ['06', '08', '10', '12', '14', '16', '18', '20']
    no_rain = random.random()
    time_of_day_randomiser = random.randint(0, 7)
    if no_rain < 0.3:
        rain_randomiser = 0
        wet_road_randomiser = 0
    else:
        rain_randomiser = random.random()
        wet_road_randomiser = random.random()
    client.simSetWeatherParameter(airsim.WeatherParameter.Rain, rain_randomiser)
    client.simSetWeatherParameter(airsim.WeatherParameter.Roadwetness, wet_road_randomiser)
    client.simSetTimeOfDay(True, "2018-11-27 " + possible_times[time_of_day_randomiser] + ":00:00")
    weather_time_dict = {'Rain': rain_randomiser, 'Roadwetness': wet_road_randomiser,
                         'Time': possible_times[time_of_day_randomiser]}
    return weather_time_dict

def random_reset(client):
    reset_randomiser = random.random()
    if reset_randomiser < 0.1:
        client.reset()
        randomise_traffic(client)

def randomise_traffic(client):
    client.simPause(False)
    initial_sim_randomisation = random.randint(0, 30)
    time.sleep(initial_sim_randomisation)
    client.simPause(True)

def generate_images(client):
    i = random.randint(0, 2)  # In AirSim 0, 1, 2 correspond to front centre, front right and front left cameras
    images_to_get = [airsim.ImageRequest(i, airsim.ImageType.Scene),
                     airsim.ImageRequest(i, airsim.ImageType.Segmentation)]
    responses = client.simGetImages(images_to_get)
    return responses, i

def save_image(img, orig_seg_string, image_directory, img_timestamp):
    if img.width == 480:
        airsim.write_file(os.path.normpath(image_directory + orig_seg_string + img_timestamp + '.png'),
                          img.image_data_uint8)
    else:
        img = Image.open(io.BytesIO(img.image_data_uint8))
        if 'orig' in orig_seg_string:
            img = img.resize((480, 360), resample=Image.BILINEAR)
        img.save(os.path.normpath(image_directory + orig_seg_string + img_timestamp + '.png'))

def save_images(image_responses, image_directory):
    assert(len(image_responses) == 2)
    orig_img = image_responses[0]
    seg_img = image_responses[1]
    img_timestamp = str(time.time()).replace('.', '')
    save_image(orig_img, '/original_', image_directory, img_timestamp)
    save_image(seg_img, '/segmented_', image_directory, img_timestamp)
    return img_timestamp

def collect_and_save_images(airsim_poses, client, image_directory, initial_rng_state):
    client.simEnableWeather(True)
    randomise_traffic(client)
    img_info = {'Initial RNG State': initial_rng_state}
    for i in range(int(args.get('--n_positions'))):
        if i % 50 == 0:
            logging.info("{}/{} positions have been visited".format(i, args.get('--n_positions')))
        random_reset(client)
        pose, collision_info, weather_time_dict = random_spawn(airsim_poses, client)
        if not collision_info.has_collided:
            responses, camera = generate_images(client)
            timestamp = save_images(responses, image_directory)
            timestamp_dict = {'Pose': pose, 'Camera': camera}
            timestamp_dict.update(weather_time_dict)
            img_info[timestamp] = timestamp_dict
        if len(img_info) >= 100:
            information_yaml(img_info, image_directory)
            img_info = {}
    return img_info

def information_yaml(img_info, image_directory):
    yaml_file_path = image_directory + '/image_information_' + time.strftime('%d-%m-%y_%H-%M') + '.yml'
    with open(yaml_file_path, 'w+') as f:
        yaml.dump(img_info, f)

def main(args):
    initial_rng_state = random.getstate()
    positions_dir = args.get('POSITIONS_DIR')
    image_directory = args.get('DATA_DIR')
    if not os.path.isdir(image_directory):
        os.mkdir(image_directory)
    airsim_poses = retrieve_coordinates_list(positions_dir)
    client = airsim.CarClient()
    client.confirmConnection()
    img_info = collect_and_save_images(airsim_poses, client, image_directory, initial_rng_state)
    information_yaml(img_info, image_directory)

if __name__ == '__main__':
    args = docopt(__doc__)
    logging.basicConfig(format="%(message)s", level=logging.INFO)
    main(args)
```

And the settings.json:

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Car",
  "ViewMode": "NoDisplay",
  "Recording": {
    "Cameras": [
        { "CameraName": "0", "ImageType": 0, "PixelsAsFloat": false, "Compress": true }
    ]
  },
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1920,
        "Height": 1440,
        "TargetGamma": 2.0
      },
      {
        "ImageType": 5,
        "Width": 1920,
        "Height": 1440
      }
    ]
  }
}
```

msb336, 2019-06-07:
@sophiefsadler can you post your debug log as well? 
Will be in `{Project Folder}/{Name of Project}/Saved/Logs`

sophiefsadler, 2019-06-14:
```
Log file open, 06/07/19 13:27:42
LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
LogWindows: File 'aqProf.dll' does not exist
LogWindows: Failed to load 'VSPerf140.dll' (GetLastError=126)
LogWindows: File 'VSPerf140.dll' does not exist
LogWindows: Failed to load 'VtuneApi.dll' (GetLastError=126)
LogWindows: File 'VtuneApi.dll' does not exist
LogWindows: Failed to load 'VtuneApi32e.dll' (GetLastError=126)
LogWindows: File 'VtuneApi32e.dll' does not exist
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(-853371005) SRandInit(-853371004).
LogTaskGraph: Started task graph with 5 named threads and 8 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.168462
LogD3D11RHI: Aftermath initialized
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin RawInput
LogPluginManager: Mounting plugin AirSim
LogPluginManager: Mounting plugin NativizedAssets
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin HTML5Networking
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin ActorSequence
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin AppleARKit
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin WindowsMoviePlayer
LogInit: Using libcurl 7.47.1
LogInit:  - built for x86_64-pc-win32
LogInit:  - supports SSL with OpenSSL/1.0.2g
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
LogInit:      CURL_VERSION_LARGEFILE
LogInit:      CURL_VERSION_IDN
LogInit:  CurlRequestOptions (configurable via config and command line):
LogInit:  - bVerifyPeer = false  - Libcurl will NOT verify peer certificate
LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
LogInit:  - CertBundlePath = nullptr  - Libcurl will use whatever was configured at build time.
LogInit: Build: ++UE4+Release-4.18-CL-3832480
LogInit: Engine Version: 4.18.3-3832480+++UE4+Release-4.18
LogInit: Compatible Engine Version: 4.18.0-3709383+++UE4+Release-4.18
LogInit: Net CL: 3709383
LogDevObjectVersion: Number of dev versions registered: 16
LogDevObjectVersion:   Dev-Blueprints (B0D832E4-1F89-4F0D-ACCF-7EB736FD4AA2): 10
[... 1285 lines cut ...]
[2019.06.07-12.28.23:548][  0]LogPawn: Error: EnableInput can only be specified on a Pawn for its Controller
[2019.06.07-12.28.23:769][  0]LogTemp: PhysXCar
[2019.06.07-12.28.23:769][  0]LogUMG: Display: Widget Class BP_SimHUDWidget_C - Loaded Fast Template.
[2019.06.07-12.28.23:962][  0]LogTemp: Warning: WARNING! AirsimCar pawns has an empty array! Be sure to override SetAirsimCarPawns() to set AirsimCarPawns to the airsim pawns!
[2019.06.07-12.28.23:992][  0]LogLoad: Took 36.901514 seconds to LoadMap(/Game/Main_CityEnvironment)
[2019.06.07-12.28.23:996][  0]LogSlate: Took 0.003084 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoBold.ufont' (160K)
[2019.06.07-12.28.24:534][  0]LogSlate: Took 0.002387 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoRegular.ufont' (155K)
[2019.06.07-12.28.24:574][  0]LogLoad: (Engine Initialization) Total time: 42.07 seconds
[2019.06.07-12.28.24:574][  0]LogExternalProfiler: No external profilers were discovered.  External profiling features will not be available.
[2019.06.07-12.28.24:600][  0]LogRawInputWindows: Warning: Found device 0 devices
[2019.06.07-12.28.24:600][  0]LogRawInputWindows: Warning: Found device 0 devices
[2019.06.07-12.28.24:600][  0]LogRawInputWindows: Warning: Device was registered succesfully but not connected (Usage:4 UsagePage:1)
[2019.06.07-12.28.24:600][  0]LogRawInputWindows: Warning: Found device 1 devices
[2019.06.07-12.28.24:600][  0]LogRawInputWindows: Warning: Device was registered succesfully but not connected (Usage:5 UsagePage:1)
[2019.06.07-12.28.24:671][  0]LogMaterial: Warning: Material /Game/PolymodelsCarsV02/Materials/m_interior.m_interior missing bUsedWithInstancedStaticMeshes=True! Default Material will be used in game.
[... cut ...]
[2019.06.07-12.28.24:673][  0]LogContentStreaming: Texture pool size now 1000 MB
[2019.06.07-12.28.25:778][  1]LogInput: Error: Duplicate mapping of key Right for axis MoveRight
[2019.06.07-12.28.25:778][  1]LogInput: Error: Duplicate mapping of key Left for axis MoveRight
[2019.06.07-12.28.25:810][  1]LogTemp: Warning: ped manager could not get any airsim car pawns!!!
[2019.06.07-12.28.39:475][331]LogRenderer: Reallocating scene render targets to support 1920x1440 Format 10 NumSamples 1 (Frame:1).

```

sophiefsadler, 2019-06-14:
I believe this is the correct debug log:

```
Log file open, 06/07/19 13:27:42
LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
LogWindows: File 'aqProf.dll' does not exist
LogWindows: Failed to load 'VSPerf140.dll' (GetLastError=126)
LogWindows: File 'VSPerf140.dll' does not exist
LogWindows: Failed to load 'VtuneApi.dll' (GetLastError=126)
LogWindows: File 'VtuneApi.dll' does not exist
LogWindows: Failed to load 'VtuneApi32e.dll' (GetLastError=126)
LogWindows: File 'VtuneApi32e.dll' does not exist
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(-853371005) SRandInit(-853371004).
LogTaskGraph: Started task graph with 5 named threads and 8 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.168462
LogD3D11RHI: Aftermath initialized
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin RawInput
LogPluginManager: Mounting plugin AirSim
LogPluginManager: Mounting plugin NativizedAssets
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin HTML5Networking
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin ActorSequence
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin AppleARKit
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin WindowsMoviePlayer
LogInit: Using libcurl 7.47.1
LogInit:  - built for x86_64-pc-win32
LogInit:  - supports SSL with OpenSSL/1.0.2g
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
LogInit:      CURL_VERSION_LARGEFILE
LogInit:      CURL_VERSION_IDN
LogInit:  CurlRequestOptions (configurable via config and command line):
LogInit:  - bVerifyPeer = false  - Libcurl will NOT verify peer certificate
LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
LogInit:  - CertBundlePath = nullptr  - Libcurl will use whatever was configured at build time.
LogInit: Build: ++UE4+Release-4.18-CL-3832480
LogInit: Engine Version: 4.18.3-3832480+++UE4+Release-4.18
LogInit: Compatible Engine Version: 4.18.0-3709383+++UE4+Release-4.18
LogInit: Net CL: 3709383
[... 1287 lines cut ...]
[2019.06.07-12.28.23:548][  0]LogPawn: Error: EnableInput can only be specified on a Pawn for its Controller
[2019.06.07-12.28.23:769][  0]LogTemp: PhysXCar
[2019.06.07-12.28.23:769][  0]LogUMG: Display: Widget Class BP_SimHUDWidget_C - Loaded Fast Template.
[2019.06.07-12.28.23:962][  0]LogTemp: Warning: WARNING! AirsimCar pawns has an empty array! Be sure to override SetAirsimCarPawns() to set AirsimCarPawns to the airsim pawns!
[2019.06.07-12.28.23:992][  0]LogLoad: Took 36.901514 seconds to LoadMap(/Game/Main_CityEnvironment)
[2019.06.07-12.28.23:996][  0]LogSlate: Took 0.003084 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoBold.ufont' (160K)
[2019.06.07-12.28.24:534][  0]LogSlate: Took 0.002387 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/RobotoRegular.ufont' (155K)
[2019.06.07-12.28.24:574][  0]LogLoad: (Engine Initialization) Total time: 42.07 seconds
[2019.06.07-12.28.24:574][  0]LogExternalProfiler: No external profilers were discovered.  External profiling features will not be available.
[2019.06.07-12.28.24:600][  0]LogRawInputWindows: Warning: Found device 0 devices
[2019.06.07-12.28.24:600][  0]LogRawInputWindows: Warning: Found device 0 devices
[2019.06.07-12.28.24:600][  0]LogRawInputWindows: Warning: Device was registered succesfully but not connected (Usage:4 UsagePage:1)
[2019.06.07-12.28.24:600][  0]LogRawInputWindows: Warning: Found device 1 devices
[2019.06.07-12.28.24:600][  0]LogRawInputWindows: Warning: Device was registered succesfully but not connected (Usage:5 UsagePage:1)
[2019.06.07-12.28.24:671][  0]LogMaterial: Warning: Material /Game/PolymodelsCarsV02/Materials/m_interior.m_interior missing bUsedWithInstancedStaticMeshes=True! Default Material will be used in game.
[2019.06.07-12.28.24:671][  0]LogMaterial: Warning:      The material will recompile every editor launch until resaved.
[... cut ...]
[2019.06.07-12.28.24:673][  0]LogContentStreaming: Texture pool size now 1000 MB
[2019.06.07-12.28.25:778][  1]LogInput: Error: Duplicate mapping of key Right for axis MoveRight
[2019.06.07-12.28.25:778][  1]LogInput: Error: Duplicate mapping of key Left for axis MoveRight
[2019.06.07-12.28.25:810][  1]LogTemp: Warning: ped manager could not get any airsim car pawns!!!
[2019.06.07-12.28.39:475][331]LogRenderer: Reallocating scene render targets to support 1920x1440 Format 10 NumSamples 1 (Frame:1).

```

msb336, 2019-06-14:
Very interesting...looks like unreal is getting some overlapping keybindings that are causing a conflict:
```
[2019.06.07-12.28.25:778][  1]LogInput: Error: Duplicate mapping of key Right for axis MoveRight
[2019.06.07-12.28.25:778][  1]LogInput: Error: Duplicate mapping of key Right for axis MoveRight
```

This is from the v1.2.2 CityEnviron, correct?

sophiefsadler, 2019-06-21:
Sorry I missed this message - I can't remember for this specific run, but I've had this issue with both v1.2.1 and v1.2.2

closed 2021-08-30 by jonyMarino