microsoft/AirSim #2009: API errors in ROS wrapper

didwjdrl12, 2019-06-10:
Hi, I am trying to use ROS wrapper with AirSim simulator Unreal engine on Linux, but  

Exception raised by the API:
rpclib: function 'getMultirotorState' (called with 1 arg(s)) threw an exception. The exception contained this information: Vehicle API for 'ComputerVision' is not available. This could either because this is simulation-only API or this vehicle does not exist.
error

This error occurs permanently, and any topic and service do not work. 
How can I solve this problem?

msb336, 2019-06-10:
It looks like you are operating in computer vision mode. getMultirotorState can only be called on a multirotor vehicle. Can you copy your settings file into the thread?

didwjdrl12, 2019-06-11:
Thank you for your reply.
This is my setting.json file.

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor"
}

I set the simmode as Multirotor, I don't know why the ROS wrapper recognize the simmode as ComputerVision

machenxiang, 2019-07-26:
> Thank you for your reply.
> This is my setting.json file.
> 
> {
> "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
> "SettingsVersion": 1.2,
> "SimMode": "Multirotor"
> }
> 
> I set the simmode as Multirotor, I don't know why the ROS wrapper recognize the simmode as ComputerVision

 I have the same problem. did you solve it?

madratman, 2019-11-04:
Please add a "vehicle" clause in your settings.json. you can look at the ones provided here https://github.com/microsoft/AirSim/tree/master/ros/src/airsim_tutorial_pkgs/settings

closed 2020-02-14 by ironclownfish

jusevitch, 2020-07-12:
Sorry to bring up this issue again, but I'm getting the same error despite having a "Vehicles" clause in my settings.json. Here is my settings.json file:

```
{
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles": {
        "Drone0": {
            "Pitch": 0,
            "Roll": 0,
            "VehicleType": "SimpleFlight",
            "X": 0.0,
            "Y": 0.0,
            "Yaw": 0,
            "Z": 0
        }
    }
}
```

Here is the error I receive when launching the airsim_node.launch file in ROS:

```
error
Exception raised by the API:
rpclib: function 'getMultirotorState' (called with 1 arg(s)) threw an exception. The exception contained this information: Vehicle API for 'ComputerVision' is not available. This could either because this is simulation-only API or this vehicle does not exist.
```

No messages are published on any of the topics.

Additional details:
* I'm using AirSim built from source on Windows 10 inside the Unreal Engine as described by [these instructions](https://microsoft.github.io/AirSim/build_windows/).
* I'm running ROS Noetic on WSL 2, Ubuntu 20.04 distribution.

cthoma20-arc, 2020-07-20:
I am having the same issue. I copied the settings.json verbatim from [here](https://github.com/microsoft/AirSim/blob/master/ros/src/airsim_tutorial_pkgs/settings/front_stereo_and_center_mono.json). Running AirSim on Windows 10, with AirSim ROS node running on ROS melodic on Ubuntu 18.04. Any help would be greatly appreciated! Thanks!

jusevitch, 2020-07-23:
The solution above works for me. When using WSL 2 on Windows 10 while running AirSim from the Unreal Engine, launching `airsim_node.launch` from WSL 2 only works when there exist two copies of the _exact same_ `settings.json` file in both `Documents\Airsim` in Windows and `~/Documents/AirSim` in the WSL 2 file system. If the settings file is missing from either location, I get an error when launching `airsim_node.launch`.

Given the growing popularity of using ROS in WSL 2, this would probably be worth either putting in the documentation or fixing.