microsoft/AirSim #2017: I want to know the distance from drone to tree, building from the depth image?

machenxiang, 2019-06-13:
[image]
Hello every one.I want to know the distance from drone to tree, building,...
What do these gray values represent the true distance? Thanks in advance.

Hezzzzzy, 2019-06-13:
you should use the float value from pfm file

machenxiang, 2019-06-13:
> you should use the float value from pfm file

Do you mean the pixel value of pfm file represent the distance between the drone and the tree?

Hezzzzzy, 2019-06-14:
> > you should use the float value from pfm file
> 
> Do you mean the pixel value of pfm file represent the distance between the drone and the tree?

yes

machenxiang, 2019-06-16:
> > > you should use the float value from pfm file
> > 
> > 
> > Do you mean the pixel value of pfm file represent the distance between the drone and the tree?
> 
> yes
Thank you very much.

msb336, 2019-06-17:
Closing thread because it looks like your issue has been solved. If you are still having problems related to this issue, please reopen the thread.

closed 2019-06-17 by msb336