microsoft/AirSim #2018: SITL with QgroundControl and px4

bwassim, 2019-06-13:
Hello everyone, 
My goal is to run Qground control with PX4 as SITL with Unreal Landscape Mountain as a simulation environment. This part is not detailed in the documentation.  I would appreciate any help on this. Thanks

- Has anybody been able to setup and get it to work?
I am running this on a Windows 10 machine with a linux terminal.  When I follow the instructions on this link https://airsim-fork.readthedocs.io/en/latest/px4_sitl.html I get this line at the end
**INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..**
- When I start the simulation on the Unreal editor, I don't see anything happening on the px4 console?
- It seems PX4 is not connected to Unreal editor simulation. 
- You can find below the json file for AirSim settings

sudo ./build/posix_sitl_default/px4 ./posix-configs/SITL/init/ekf2/iris
commands file: ./posix-configs/SITL/init/ekf2/iris

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [dataman] Unknown restart, data manager file 'rootfs/fs/microsd/dataman' size is 11405132 bytes
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..

json settings file for unreal project
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "vehicles": {
  	"PX4": {
  		 "VehicleType": "PX4Multirotor",
  		 "SitlIp": "",
  		 "UdpIp": "127.0.0.1",
  		 "UdpPort": 14560,
  		 "UseSerial": false
  	}  
  }
}

------------------------------------

dhost88, 2019-06-26:
I am having the same exact issue. I can control the Multirotor in the mountain simulation with a joystick, but it is the UDP communication between QGC and Unreal that is the issue.

Any help would be appreciated

HanbumKo, 2019-07-22:
bwassim,
Did you run the Unreal environment?

bwassim, 2019-07-25:
Yes of course. I was also unable to make the PX4 work in an Ubuntu 16.04 with the PX4 v1.8.2 version. It runs but the Unreal environment does not seem to connect to it. Is anyone having any issue with the PX4?

chris-aeviator, 2019-11-08:
I think in general the docs should go deeper into how settings things up - I'm struggling to understand the settings for PX4 SITL - could someone please go deeper into what's the settings

`QgcPort`, `SitlPort` & `UDP` port?

My setup is
- airsim binary (linux)
- px4-sitl via the official docker container, port 14556 (UDP) shared with the host - (`make px4_sitl_default none_irirs` log saying `waiting for the simulation to connect on TCP 4560`) 
- QGroundcontrol Setting is a mistery to me, whenever I chose the 14556 it says Port already in use (from docker, sure)

xlong0513, 2019-11-22:
I have the same question. Did anyone fix it? Appreciate it!

jonyMarino, 2020-11-23:
Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing an issue
*Attach a screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-11-23 by jonyMarino