microsoft/AirSim #2020: Run LandscapeMountains.sln failed.

dengzq186, 2019-06-14:
I was trying to run arisim in UE, i worked step by step as the doc said,When i run LandscapeMountains.sln(step 9.Press F5 to run. This will start the Unreal Editor. ) i got an error in VS,ignore the chinese.
When i used VS2015 to download the AirSim source code from git, it told me in terminal UI that i should update it to VS2017 when i ran the build.cmd.  
Someone says this doc is only for VS2015, i tried to use VS2015 to open the LandscapeMountains.sln, but VS showed Compatibility error just like we used the lower VS to open the higher VS project.
Current my environment is VS2017, UE version 4.22.1.
Did i forgot something, someone helps me, thanks a lot...

VS error as below:
1>Creating makefile for LandscapeMountainsEditor (no existing makefile)
1>Parsing headers for LandscapeMountainsEditor
1>  Running UnrealHeaderTool "D:\AirSim\UE\LandscapeMountains\LandscapeMountains\LandscapeMountains.uproject" "D:\AirSim\UE\LandscapeMountains\LandscapeMountains\Intermediate\Build\Win64\LandscapeMountainsEditor\DebugGame\LandscapeMountainsEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
1>Reflection code generated for LandscapeMountainsEditor in 6.6148369 seconds
1>Using Visual Studio 2017 14.11.25503 toolchain (D:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Tools\MSVC\14.11.25503) and Windows 10.0.16299.0 SDK (C:\Program Files (x86)\Windows Kits\10).
1>Building 12 actions with 8 processes...
1>  [1/12] Default.rc2
1>  [2/12] Default.rc2
1>  [3/12] SharedPCH.Engine.NonOptimized.cpp
1>d:\program files\epic games\ue_4.22\engine\source\runtime\core\public\Windows/WindowsPlatformCompilerSetup.h(21): error C2338: Visual Studio 2017 version 15.6 is required to compile on Windows. Please install updates through the Visual Studio installer.
1>D:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets(44,5): error MSB3075: 命令“"D:\Program Files\Epic Games\UE_4.22\Engine\Build\BatchFiles\Build.bat" LandscapeMountainsEditor Win64 DebugGame -Project="D:\AirSim\UE\LandscapeMountains\LandscapeMountains\LandscapeMountains.uproject" -WaitMutex -FromMsBuild”已退出(exit)，代码(code)为 5。请验证您是否拥有运行此命令的足够权限。(Please verify that you have sufficient permissions to run this command)
1>已完成生成项目“LandscapeMountains.vcxproj”的操作 - 失败Failed。
========== 生成: 成功 0 个，失败 1 个，最新 0 个，跳过 1 个 ==========

LandscapeMountains.uproject as below, the ogrinal EngineAssociation is 4.22, so i keep it.
{
	"FileVersion": 3,
	"EngineAssociation": "4.22",
	"Category": "Samples",
	"Description": "",
	"Modules": [
		{
			"Name": "LandscapeMountains",
			"Type": "Runtime",
			"LoadingPhase": "Default",
			"AdditionalDependencies": [
				"AirSim"
			]
		}
	],
	"TargetPlatforms": [
		"MacNoEditor",
		"WindowsNoEditor"
	],
	"Plugins": [
		{
			"Name": "AirSim",
			"Enabled": true
		}
	]
}

dengzq186, 2019-06-14:
Additional i repeated to run build.cmd in VS2015, It showed:
Hello there! We just upgraded AirSim to Unreal Engine 4.18 and Visual Studio 2017. Here are few easy 
steps for upgrade so everything is new and shiny: 
https://github.com/Microsoft/AirSim/blob/master/docs/unreal_upgrade.md.

And this is the first time i tried to build AirSim, never happened in VS2015 before

msb336, 2019-06-17:
@dengzq186 It looks like you are using SDK 10.0.16299.0 try retargeting your solution to 8.1 and building again

dengzq186, 2019-06-18:
> 
> 
> @dengzq186 It looks like you are using SDK 10.0.16299.0 try retargeting your solution to 8.1 and building again

Thanks for your reply.

msb336, 2019-06-18:
@dengzq186 Did you retarget the solution to the proper SDK? Visual Studio version and compiler SDK are not the same thing.

To retarget AirSim to 8.1 SDK:
1. Open the solution in VS
1. Right click the solution in Solution Explorer
1. Select `Retarget Solution`
1. Change `Windows SDK Version` to `8.1`

closed 2019-09-25 by msb336