microsoft/AirSim #2021: Changing AirsimGamemode Controller to custom (blueprint) controller

nalfano16, 2019-06-14:
OS: Windows
AirSim Version: 1.2
Unreal Version: 4.22

I am attempting to create an autonomously driving vehicle using airsim. I have already scripted a pretty intense blueprint algorithm that does every I need it too, when I  choose "Airsim Gamemode" the "selected gamemode" tabs become greyed out and I cannot change the "Player Controller Class" I am more familiar with blueprint scripting than C++. I am assuming I need to edit the AirsimGamemode.cpp file? I also would like to change to a custom vehicle, where I am encountering the same error. Any help on getting this changed as desired would be helpful. Thanks.

nalfano16, 2019-06-17:
I should add, the autonomous driving blueprint is a player controller class

msb336, 2019-06-17:
@nalfano16 

Yes, you will need to modify AirSimGameMode to add a new controller class, but this will most likely break most -- if not all -- of the AirSim API. AirSim doesn't make any active decisions on the server side, it only listens for commands sent from the client. It's very unusual to cook the policy and API controls of your vehicles into an unreal binary (if you wanted to tweak your policy you would have to rebuild the entire project binary!).

If you are blueprinting an autonomous vehicle, you most likely are also handling the physics dynamics of your vehicle outside of AirSim. Perhaps creating your own game mode outside of AirSim is a simpler solution to your problem.

nalfano16, 2019-06-18:
@msb336 I was put on this project with little to no knowledge about either of unreal or AirSim so I have been learning as I go so forgive my ignorance. Without using the AirSimgamemode am I still taking advantage of using AirSim? I was told to use it but don’t know all of what it provides.

msb336, 2019-06-18:
No, you won't be taking advantage of airsim, but you most likely will not be taking advantage of airsim if you modify your fork to use blueprints instead of the API.

The airsim client-side API is very intuitive, and I highly recommend taking the time to learn it (Python will be easier than c++). If you are truly set on building all of the functionality in blueprints, AirSim is definitely not the ideal plugin for you.

There are some python demos [here](https://github.com/microsoft/AirSim/tree/master/PythonClient), and you can implement your own vehicle types using [pawn paths](https://github.com/microsoft/AirSim/blob/master/docs/settings.md#pawnpaths).

Pawn paths will require some delving into unreal engine, as you will need to build your own vehicle pawn (which it seems like you might already have).

I hope this helped you on your quest!

nalfano16, 2019-06-18:
@msb336, Alright so one final thing to make sure my understanding is correct. What you recommend is to take the functionality of the blueprint I created and, using python, write an API?

msb336, 2019-06-18:
You most likely be able to use the APIs already in existence to suite your needs.

You should convert your blueprint into a Python or C++ scripts that uses the airsim client to make calls to the simulator.

Hello Car [python](https://github.com/microsoft/AirSim/blob/master/PythonClient/car/hello_car.py)/[c++](https://github.com/microsoft/AirSim/blob/master/HelloCar/main.cpp) is a good place to start

closed 2019-06-20 by msb336