microsoft/AirSim #2022: Airsim City Environment not starting after Windows update

MartinAFHWS, 2019-06-17:
Hi,
I've successfully build Unreal Engine and Airsim on Windows 10 and would like to use the city environment with moving vehicles and pedestrians for a drone (px4-SITL) algorithm to avoid crashing into the moving vehicles and pedestrians. Before the last windows update i could launch the city environment 1.2.0 - 1.2.2 but now it doesnt start. It gets to a loading screen which is completly black(which is normal) and then just shuts down. There is no Error Message popping up. 

I already tried to start other versions of the city environment and the blocks environment but none of these are starting correctly. I also tried to use different ue4 versions which also didnt work.

unreal: 4.18 and 4.22

Logfile of City environment 1.2.2:
[CityEnviron-backup-2019.06.14-13.49.52.log](https://github.com/microsoft/AirSim/files/3297624/CityEnviron-backup-2019.06.14-13.49.52.log)

Thanks

ironclownfish, 2019-06-17:
Hi Martin.
Have you tried the solution suggested here?
https://answers.unrealengine.com/questions/758236/all-my-plugins-nodes-get-call-unknown-function-err.html

The issue was resolved by clearing the contents of C:/Users/{system username}/AppData/Local/UnrealEngine.

I was able to run the city environment without issue. If the above fix doesn't work, can you let us know which exact Windows 10 version you're using?

MartinAFHWS, 2019-06-18:
Unfortunately it didn't work.
My Windows version:

Edition: Win 10 Pro
Version 1809
OS Build: 17763.557

ironclownfish, 2019-06-18:
I'm afraid it must be a local dependency issue. We're able to run the latest environments (including city) on Win 10 v1809 - 17763.557

MartinAFHWS, 2019-06-25:
Thank you for your help.

I found the issue. It was a faulty airsim settings.json file, which somehow didnt show up in the log file.

closed 2019-06-25 by MartinAFHWS

AhmadRamezAli, 2025-04-28:
when i try to drive the car, car ignore me like my car isn't exist. i dont know what is the problem. any one have the full city environ package, because i found cityFnviron.001 and 002 but i didnt find .003