microsoft/AirSim #2024: What's the relationship between throttle and speed?

Hezzzzzy, 2019-06-18:
hi guys
when I control the car by python api, I need input a value named "throttle",  then I will get a linear speed,  but I found it's not "1:1", anyone knows the conversion relationship??

thx a lot

msb336, 2019-06-18:
I believe the car's velocity is modeled as a first or second order differential equation, where throttle is an input to that equation. If you set throttle to a unit step input and record velocity over time, you will see something like [this](https://lpsa.swarthmore.edu/Transient/TransStep/imgF2.gif), except v(t) will converge on max velocity, which will be greater than 1

Hezzzzzy, 2019-06-19:
> I believe the car's velocity is modeled as a first or second order differential equation, where throttle is an input to that equation. If you set throttle to a unit step input and record velocity [...]

thx for your answer, I also have a question,  is there any way  can control the car by a constant scalar speed？

Hezzzzzy, 2019-06-20:
@msb336 I found a piece of c++ code
`    // Engine 
    // Torque setup
    movement->EngineSetup.MaxRPM = 5700.0f;
    movement->EngineSetup.TorqueCurve.GetRichCurve()->Reset();
    movement->EngineSetup.TorqueCurve.GetRichCurve()->AddKey(0.0f, 400.0f);
    movement->EngineSetup.TorqueCurve.GetRichCurve()->AddKey(1890.0f, 500.0f);
    movement->EngineSetup.TorqueCurve.GetRichCurve()->AddKey(5730.0f, 400.0f);` 
And I think this is control the acc of car, but I am the new of the "torque curve", I don't know how to set the curve to get the faster acc,  can you help me? thx guys

msb336, 2019-06-20:
The car dynamics are not part of airsim's code, so I would suggest posting your question on the [unreal forums](https://forums.unrealengine.com/).

Base class for ACarPawn is defined [here](https://api.unrealengine.com/INT/API/Plugins/PhysXVehicles/AWheeledVehicle/index.html), engine setup defined [here](https://api.unrealengine.com/INT/API/Plugins/PhysXVehicles/FVehicleEngineData/index.html)

closed 2019-06-20 by msb336