microsoft/AirSim #2025: how to reset has_collided flag

wtq1996, 2019-06-18:
Hi,

I want to know how to reset the  has_collided flag without using client.reset(). Cause I want to set the initial position of the car randomly using API after detecting a collision while calling client.reset() would always start at the same position.

wtq1996, 2019-06-18:
I think I got a clue, I can just simply set car position after calling reset(), but still wondering if there is a way to reset the has_collided flag

msb336, 2019-06-18:
Unfortunately there is no API for that. Effort-wise, it would be easiest to just set a random vehicle position using `simSetVehiclePose()` after resetting. If you are deadset on being able to toggle the has_collided flag, you will need to build that API functionality.

wtq1996, 2019-06-20:
@msb336 Hi, is it possible to read the car's current control input using API? LIke the number which shows in the left part of the window. I am trying to do imitation learning now, so I want to use the xbox controller and keyboard to record my control policy. I used a third party library to read the Xbox controller's input. But the problem is that the Xbox controller cannot move the car backward (at least I didn't figure out how to do that). Although I can use the keyboard simultaneously (the down arrow) to make the car move backward, there seems to have an unsolvable problem which is that if I want my keyboard press detecting code  to work I need to click the command window in which I run the .py file,
in that case, the down arrow input won't give to the car.
So, my question is, how to make the car move backward using Xbox controller or read the car's received control directly?

msb336, 2019-09-25:
For your first question, yes, there is a `getCarControls` api available. for the second question, the keybinding for moving backwards is the down arrow. Issues you have with mapping from xbox control to keypresses are unrelated to airsim, and should be directed towards the proper forms.

closed 2019-09-25 by msb336