microsoft/AirSim #2026: client.simGetImages is so slow

elmtalab, 2019-06-18:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

msb336, 2019-06-18:
@elmtalab 

This is a core functionality issue that is still a work in progress. Image capture from a camera requires Unreal to render a new viewpoint in the scene. The higher the image resolution, the longer it takes for unreal to render the extra image.

As a short term solution, you will be able to boost your frame rate by reducing the resolution of the image you are trying to capture.

elmtalab, 2019-06-18:
I just need it to run offline like what OpenAI gym environments provide. I want to collect dataset. 
is there anyway to stop image rendering?

hipforth, 2021-11-22:
ClockSpeed set 0.1666 is useful.