microsoft/AirSim #2028: changing drone pawn to use in built package for UE4?

JackGit3, 2019-06-19:
In the settings JSON I've specified a custom pawn to use in the editor and this works fine, however when I package and run the resulting exe I'll get an error stating it failed to find my custom pawn. I read the airsim guide and it says to make sure the bp is cooked but it is in my content folder and from UE4 docs it seems everything in content folder is cooked by default.

Any ideas on how I can get the packaged build to use my custom pawn?

JackGit3, 2019-06-20:
Fixed, had to add the directory under additional directories to cook in packaging settings after all.

closed 2019-06-20 by JackGit3

clev0910, 2020-05-14:
Hello, I'd like to use a custom drone in place of the existing one, and I've specified the path to it in the settings too, but Unreal Engine crashes when I try to run it. I don't quite understand what "cooking" is here. Could you please explain what changes need to be made for a custom drone to work?