microsoft/AirSim #2029: The location of the propeller is moved.

gniwon, 2019-06-20:
I am using my default model(BP_FlyingPawn), and recently I found that the propellers would shift a little after a long flight. What is the problem? Is there solution?

wuming0624, 2019-11-12:
I noticed the same issue. Mine shifted 45 degree after a long run, and started to block my camera. Any fix?

jonyMarino, 2020-08-17:
Closed due to incompleteness.
Please feel free to open a new issue if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-08-17 by jonyMarino