microsoft/AirSim #2033: Creating a line visible in the segmentation, but not the rendered image

machampagne1, 2019-06-21:
hi, I'm trying to create a visible line in the segmentation, but I need this line to *not* be visible in the normal render of the scene. 

My idea was to put down a cube and give it a transparent texture, but AirSim won't see it !

How do I fix this ?

Thanks !

saihv, 2019-10-09:
The segmentation image uses Unreal object IDs as the means to generate a 'palette' of colors. If you are able to create a line with a transparent texture, perhaps you can try using `simSetSegmentationID` on that particular object to give it a specific color. Please see PythonClient/computer_vision/segmentation.py for better insight.

closed 2020-01-20 by saihv

chickenbestlover, 2020-11-27:
I have tried @saihv's suggestion and found out that Airsim API does not create any segmentation label for transparent meshes.