microsoft/AirSim #2035: How to set the same color to all cars/buildings... in semantic segmentation images?

petervlad, 2019-06-22:
[image]

Is it possible to do this from settings.json file?

petervlad, 2019-06-22:
Found out that you can't. By setting the MeshNamingMethod as "StaticMeshName" in Segmentation Settings the colors will still be different but will be the same and not differ each time you open the environment.

closed 2019-06-22 by petervlad

YKembal, 2020-03-31:
Is this image from the city enviroment ? If yes , can you tell us how did you get this segmented image because when we retrieve the segmented image from airsim it doesn't include the cars and people in the image and i see here that they are detected in your photo.
If you use the segmentation image of airsim please show me your parameters and if not , can you help me find the way of producing this segmented image ?