microsoft/AirSim #2039: AirSim for Unreal Engine  4.21 or higher

hassanyf, 2019-06-26:
Will AirSim be updated to Unreal Engine 4.20, 4.21 or 4.22?

msb336, 2019-06-26:
@hassanyf current master branch is compatible with ue4.22

closed 2019-06-26 by msb336

hassanyf, 2019-06-27:
The documentation for Linux build says its available for 4.18 only. [https://microsoft.github.io/AirSim/docs/build_linux/](https://microsoft.github.io/AirSim/docs/build_linux/)

msb336, 2019-06-27:
Looks like the docs are a little out of date. Thanks for bringing this to my attention.

As of [this merge](https://github.com/microsoft/AirSim/pull/1888), Airsim is compatible with ue4.22.

thejeshk, 2020-02-21:
Hi @msb336 

I am planning to use AIRSiM plugin for my project. I have installed both Unreal versions 4.18 and 4.22. But I have created a custom Map in 4.22 version. 

Could you please confirm if I can use the 4.22 version or I should create different map in 4.18 to use AIRSIM plugins

Thanks and regards, 
Thejesh

DavidWWalker, 2021-03-29:
AirSim and UnrealEngine v4.25.4. 

There is no compatible vehicle created  for current SimMode.

settings.json file copies directly from website on how to setup AirSim.

QGroundStation reports GPS  is unhealthy. Will not permit me to arm the simulator.  Tried Guided and Guided without GPS modes

Sim model on Taranis Plus loaded and joystick coordinates displayed in simulator.

Would appreciate any steps I can take to overcome these issues!