microsoft/AirSim #2040: Live drone deployment with Mavlink proxy issues

delphifissure, 2019-06-27:
I'm trying to simply connect a live PX4 drone to DroneShell so I can run a python script. The drone is an Intel Aero running Ubuntu 16.04.6 LTS and Airlib is installed. I'm on a Windows 10 laptop. For the below I'm keeping Unreal and QGroundControl out of the mix. 

First I open three terminals on my laptop and ssh into the drone in each.
1. Start the MavLinkTest proxy. The parameters load and I can arm / disarm etc., but the messages aren't forwarding.
```
aero@aero:~/AirSim/build_debug/output/bin$` ./MavLinkTest -serial:/dev/ttyS1,115200 -logdir:. -proxy:127.0.0.1:14560
Connecting to serial port /dev/ttyS1, baudrate=115200
Connecting to UDP Proxy address 127.0.0.1:14560
Downloading drone parameters so we know how to control it properly...
MavLinkConnectionImpl: Error handling message 30 on connection 'drone', details: MavLinkConnectionImpl: Error sending message on connection 'proxy', details: UdpClientPort socket send failed with error: 111
```
2. Start DroneServer. It doesn't receive any messages because the proxy isn't working...
```
aero@aero:~/AirSim/build_debug/output/bin$ ./DroneServer 0
WARNING: This is not simulation!Server connected to MavLink endpoint at 127.0.0.1:14560
Hit Ctrl+C to terminate.
Connecting to UDP port 14560, local IP 127.0.0.1, remote IP...
Connected over UDP.
Connecting to PX4 SITL UDP port 14556, local IP 127.0.0.1, remote IP...
Connected to SITL over UDP.
not receiving any messages from HIL, please restart your HIL node and try again
not receiving any messages from SITL, please restart your SITL node and try again
```
3. Start DroneShell. It connects, but doesn't respond to commands since it's not getting the messages.
```
aero@aero:~/AirSim/build_debug/output/bin$ ./DroneShell
==||=>
        Welcome to DroneShell 1.0.
        Type ? for help.
        Microsoft Research (c) 2017.

Waiting for connection -
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
==||=>
```
4. Per [issue 283](https://github.com/microsoft/AirSim/issues/283) I tried:
```
./MavLinkTest -serial:/dev/ttyS1,115200 -logdir:. -**server**:127.0.0.1:14560
Connecting to serial port /dev/ttyS1, baudrate=115200
Connecting to UDP Server address 127.0.0.1:14560
Downloading drone parameters so we know how to control it properly...
    PX4_CUSTOM_MAIN_MODE_MANUAL
```
...but DroneServer still isn't getting any messages and so DroneShell connects but doesn't respond to commands.

My drone's settings.json:
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",    "SettingsVersion": 1.2,
    "DefaultVehicleConfig": "PX4",
    "LocalHostIp": "127.0.0.1"
    "SimMode": "Multirotor",
    "PX4": {
        "VehicleType": "PX4Multirotor",
        "LocalHostIp": "127.0.0.1"
        "UdpIp": "127.0.0.1"
        "UdpPort": 14560,
        "UseSerial": false,
        "SerialPort": "/dev/ttyS1",
        "SerialBaudRate": 115200,
        "VehicleCompID": 1,
        "VehicleSysID": 135,
        "ApiServerPort": 41451
        }
}
```

Thanks!

lovettchris, 2019-06-27:
Thanks for posting this issue.  I think you should not be using localhost ip of 127.0.0.1.  That network is limited to your machine and cannot send/receive messages to another device.  So presumably your drone is one the same wifi network as your laptop and that you have opened the firewall for mavlink ports 14550, 14560, etc.  Once you do that QGroundCOntrol should be able to connect to the drone.  If this works, then DroneServer should be able to connect also using the ip address associated with your wifi network interface, which you can find using ipconfig.  You should not need MavLinkTest as a proxy in this case.

delphifissure, 2019-06-27:
QGroundControl has no problem connecting to the drone. For ref, I also tried using my drone's IP vs. localhost:
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",    "SettingsVersion": 1.2,
    "DefaultVehicleConfig": "PX4",
    "LocalHostIp": "192.168.1.xxx"
    "SimMode": "Multirotor",
    "PX4": {
        "VehicleType": "PX4Multirotor",
        "LocalHostIp": "192.168.1.xxx"
        "UdpIp": "192.168.1.xxx"
        "UdpPort": 14560,
        "UseSerial": false,
        "SerialPort": "/dev/ttyS1",
        "SerialBaudRate": 115200,
        "VehicleCompID": 1,
        "VehicleSysID": 135,
        "ApiServerPort": 41451
        }
}
```
...but DroneServer still isn't getting any messages. I'll try again in a few hours without using MavLinkTest proxy, thanks for that idea.

lovettchris, 2019-06-27:
LocalHostIp should not be the drone's IP address.  It should be the local ip address of your laptop on the same network as the drone.  So if your local network is 192.168.1.*, then your router will show two devices, let's say 192.168.1.20 is your laptop and 192.168.1.25 is your drone, then LocalHostIp should be 192.168.1.20.  The reason this is a setting, and DroneServer doesn't just assume you want to use 192.168.1.20 is because some computers have multiple network interfaces, like a hard wired ethernet and wifi.  This setting allows you to pick which network you want to connect to.

delphifissure, 2019-06-27:
Thanks! Are you saying LocalHostIp should be set to the laptop IP in the both the laptop and drone's settings.json files? What about UdpIp?

delphifissure, 2019-06-29:
If I set my Drone's settings.json to the laptop IP for both UDP and localhost and run Droneserver, I get an exception.

```
aero@aero:~/AirSim/build_debug/output/bin$ ./DroneServer -?
WARNING: This is not simulation!
terminating with uncaught exception of type std::__1::system_error: bind: Cannot assign requested address
Aborted (core dumped)
```

closed 2021-08-30 by jonyMarino