microsoft/AirSim #2041: Px4 unreal freezes everytime when exiting the simulation

bwassim, 2019-06-27:
I am having exactly the same issue as in this link
 https://github.com/Microsoft/AirSim/issues/719#

I am on ubuntu 18.04. I have tried with with UE4.22.3 and UE4.18. For both versions, when I run LandscapeMountain on SITL, as soon as push the stop button or escape. The UEditor freezes without any message on the terminal. Any idea? Thank you

bwassim, 2019-06-28:
I tried to run the editor on from the terminal and then launch the LandscapeMountain from the Unreal editor, when I stop the simulation I get the error below 

[2019.06.28-07.10.20:508][332]LogTemp: Error: Exception occurred while updating world: MavLinkConnectionImpl: Error sending message on connection 'hil', details: UdpClientPort socket send failed with error: 111

bwassim, 2019-07-26:
I think the issue is related to Ubuntu 18.04, everything works fine with Ubuntu 16.04

bwassim, 2019-07-26:
Just realized the same issue exist on Ubuntu 16.04 too.

machenxiang, 2019-08-08:
> I think the issue is related to Ubuntu 18.04, everything works fine with Ubuntu 16.04

This problem also occurs in Ubuntu16.04

dangviethieu, 2019-12-10:
Anyone can tell me status of this issue?

saifullah3396, 2019-12-12:
Facing the same issue, but it does not seem to have anything to do with px4. The problem occurs even without connecting it with px4.

saifullah3396, 2019-12-12:
Same issue here with UseSerial=False.

harnix, 2020-03-31:
Experience the same issue, running Unreal Engine on Ubuntu 18.04

saifullah3396, 2020-03-31:
@harnix See my solution for this at https://github.com/microsoft/AirSim/issues/2112#issuecomment-567880224. You'll have to rebuild the code with a timeout on udp connection.

harnix, 2020-03-31:
@saifullah3396 Thanks for the swift response. I'll check it out now.

jonyMarino, 2020-09-02:
treated on #2112

closed 2020-09-02 by jonyMarino