microsoft/AirSim #2047: Is there any way to get normal vector under camera coordinate?

xiaoxiangyeyuwangye, 2019-07-01:
Hi, everyone!
Thanks for this awesome work.
If I understand correctly, the output normal vector is under world coordinate.
If I want to get normal vector under camera coordinate, does anyone have any suggestion about how to realize it? With Blueprint or C++?
Thanks for any suggestion!

royuqp, 2019-07-04:
Hi, you can get camera space normals by modifying BP a little!

1. Open AirSim/Contents/HUDAssets/NormalsMaterial on the UE4 Editor. As you can see in the original BP of NormalsMaterial, it will outputs world space normals.
2. Insert TransformVector Node between SceneTexture:WorldNormal node and Multiply Node. Click the TransformVector Node and set Source=World Space and Destination=View Space in the Details panel.
3. (Optional) swap/flip resulting RGB(XYZ) channels so that it matches to your camera coordinates convention.

Another subtle thing you need to make sure is that the target gamma value of the render target associated to NormalsCaptureComponent is set to be 1.0 (otherwise the output value will be exponentiated by some factor). You can do this by editing "CaptureSettings" in settings.json and explicitly setting "TargetGamma" to be 1.0 for "ImageType" 6.

Hope it'll work for you!

closed 2020-01-20 by saihv

aelassam, 2023-07-06:
Hi

I use Airsim to get normal maps for vanishing point (VP )detection. I need the normals in the camera space to calculate the vanishing points. I followed your suggestion however I don't how the coordinate axes in the view space are aligned. It's crucial for me because the VP is calculated as the cross product of the normal vector and the up vector.

Thanks for any suggestions
!