microsoft/AirSim #2051: DllNotFoundException: AirsimWrapper

Shwethakrishnaa, 2019-07-02:
Follow the airsim for windows build , the error is below!
And I can not control the car by using WASD or arrow, the car is not moving

DllNotFoundException: AirsimWrapper
AirSimUnity.VehicleCompanion.InitDelegators () (at Assets/AirSimAssets/Scripts/Vehicles/VehicleCompanion.cs:85)
AirSimUnity.VehicleCompanion..cctor () (at Assets/AirSimAssets/Scripts/Vehicles/VehicleCompanion.cs:30)
Rethrow as TypeInitializationException: The type initializer for 'AirSimUnity.VehicleCompanion' threw an exception.
AirSimUnity.Vehicle.Start () (at Assets/AirSimAssets/Scripts/Vehicles/Vehicle.cs:58)
AirSimUnity.Car.Start () (at Assets/AirSimAssets/Scripts/Vehicles/Car/Car.cs:25)

windows 10.
I also tried cloning git repo and started the step from the beginning. Still getting this error.
[image]

msolive1985, 2019-08-07:
I have the same error. Did you have any success?

Shwethakrishnaa, 2019-08-07:
Nope

mattkelsey, 2019-09-10:
Same issue, running on Arch Linux

closed 2019-09-10 by Shwethakrishnaa