microsoft/AirSim #2052: Get simulator subwindow images in parallel

mahieyin-rahmun, 2019-07-02:
Hello @sytelus and @madratman,

I have been using AirSim to follow a target. However, the *`SimGetImages`* renders pretty slow images, which results tracker to miss the target due to large pixel distance mismatch.

However, we solved this issue with slowing down the clockspeed in settings.json and in the UE simulator using slomo command.

But, the above solution results in very slow rendering of simulation, which is not feasible for any demonstration/results analysis purposes. We are now trying to use feedback from the subwindows (due to fast rendering). Is there a way we could get the subwindow feedback images directly (e.g., inter-process communication)? By subwindow I mean the highlighted boxes in red. 

[image]

jinkman, 2020-01-04:
can you tell me how to follow a target,I was very confused.

madratman, 2020-08-11:
https://github.com/microsoft/AirSim/pull/2713

yuanzihao, 2024-07-05:
I am facing the same problem, only get about ~5 FPS by simGetImage API, but the fps rendered in subwindow is what i actually need.