microsoft/AirSim #2053: Could not unzip CityEnviron.zip.001 & CityEnviron.zip.002 (v1.2.0 - Linux)

jobs-github, 2019-07-02:
* Version: v1.2.0 - Linux
* Environments: CityEnviron.zip.001, CityEnviron.zip.002

I got the following error:

    7-Zip [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
    p7zip Version 16.02 (locale=utf8,Utf16=on,HugeFiles=on,64 bits,8 CPUs x64)

    Scanning the drive for archives:
    1 file, 1782579200 bytes (1700 MiB)

    Extracting archive: CityEnviron.zip.001

    CityEnviron.zip
    ERRORS:
    Headers Error
    Unconfirmed start of archive

    CityEnviron.zip
    WARNINGS:
    There are data after the end of archive

    --
    Path = CityEnviron.zip.001
    Type = Split
    Physical Size = 1782579200
    Volumes = 2
    Total Physical Size = 3117962657
    ----
    Path = CityEnviron.zip
    Size = 3117962657
    --
    Path = CityEnviron.zip
    Type = zip
    ERRORS:
    Headers Error
    Unconfirmed start of archive
    WARNINGS:
    There are data after the end of archive
    Physical Size = 3117932051
    Tail Size = 30606

    Would you like to replace the existing file:
    Path:     ./CityEnviron
    Size:     0 bytes
    Modified: 2018-08-15 11:15:54
    with the file from archive:
    Path:     CityEnviron/Binaries/Linux/CityEnviron
    Size:     862864656 bytes (823 MiB)
    Modified: 2018-08-15 14:34:38
    ? (Y)es / (N)o / (A)lways / (S)kip all / A(u)to rename all / (Q)uit? Y

    ERROR: Data Error : CityEnviron/Content/Paks/CityEnviron-LinuxNoEditor.pak

    Sub items Errors: 1

    Archives with Errors: 1

    Warnings: 1

    Open Errors: 1

    Sub items Errors: 1

dhost88, 2019-07-02:
I have same problem on Windows 10

LapinCodeur, 2019-07-18:
I had successfully extract it with errors, but it doesn't work, it seems that you need libopenvr. However the file itself is a bit weird and need to be fixed. I think that there is no way to get it functional on Linux right now....

closed 2020-01-20 by saihv

dimikout3, 2020-02-11:
I still get the error jobs-github mentioned