microsoft/AirSim #2055: moveToPositionAsync only effect once then do nothing

vxgu86, 2019-07-03:
In HIL ,the code in DQNDrone.py,

client.moveToPositionAsync(initX, initY, initZ, 5).join()
client.moveByVelocityAsync(1, -0.67, -0.8, 5).join()

only effect once then do nothing.
it flied to the position, then landed due to the failsafe mode due to no instructions received, but the py code is doing 
client.moveByVelocityAsync(...
in the while loop

it's working well in SimpleFlight, but I want to try the HIL, did anyone ever run into this problem?

PatrickUtz, 2019-07-15:
If you are using a while loop to continuously call that function I would recommend using `moveToPosition()` instead of `moveToPositionAsync()`. This is due to the fact that the "Async()" version waits for the current process (in this case the drone moving to whatever position you told it to) to finish before executing the next available piece of code. By calling `moveToPosition()` instead, your function will send the command to Airsim/Unreal editor and return back immediately to execute the next line of code (thus your program does not get caught up in anyway). Try it out and let me know if it works!

AloshkaD, 2019-07-18:
The use of move commands without Async has been 1-deprecated and 2- does not delay the execution. .join is wat makes the command wait for execution. See this https://github.com/Microsoft/AirSim/blob/master/docs/upgrade_apis.md

closed 2020-01-20 by saihv