microsoft/AirSim #2057: How to control the drone by ROS?

machenxiang, 2019-07-03:
1. I want to control the drone fly by ROS, but I didn't find the order in ROS? Whatever thank you in advance.

ldgcug, 2019-07-11:
you can write the ros node by using the api , then Publisher topics and Sunscribers topics

machenxiang, 2019-07-12:
> you can write the ros node by using the api , then Publisher topics and Subscribers topics

但是官网给的教程很不详细，而且还有错
but the tutorial isn't very clear, and I think it even has some bugs

ldgcug, 2019-07-12:
> > you can write the ros node by using the api , then Publisher topics and Subscribers topics
> 
> 但是官网给的教程很不详细，而且还有错
> but the tutorial isn't very clear, and I think it even has some bugs

官网只提供了姿态和图像的ros程序，图像方面确实是存在一些问题，但是我们可以作为参考去修改，另外，需要自己订阅起飞降落和速度控制和位置控制等的话题，这是他们没有提供的。
还有另一种方法，你可以通过sitl来实现mavros的控制。
多看看PythonClient里面的python程序。
The official website only provides the ros program for gestures and images. There are some problems in the image aspect, but we can modify it as a reference. In addition, we need to subscribe to the topic of take-off landing and speed control and position control. This is what they did not provide. .
There is another way, you can achieve mavros control through sitl.
Take a look at the Python program inside PythonClient.

machenxiang, 2019-07-12:
> The official website only provides the ros program for gestures

对我就是没找到位置控制的话题，所以搞了很久！
Because,i don't find the the position control Topic , i take long time on this!

machenxiang, 2019-07-22:
> 官网只提供了姿态和图像的ros程序，图像方面确实是存在一些问题，但是我们可以作为参考去修改，另外，需要自己订阅起飞降落和速度控制和位置控制等的话题，这是他们没有提供的。
> 还有另一种方法，你可以通过sitl来实现mavros的控制。
> 多看看PythonClient里面的python程序。
> The official website only provides the ros program for [...]
请问一下你做过AirSim控制真飞机吗？
I'm sorry to bother you, did you use AirSim control the real drone?

ldgcug, 2019-07-22:
> you can write the ros node by using the api , then Publisher topics and Subscribers topics
>
>
> 但是官网给的教程很不详细，而且还有错
> but the tutorial isn't very clear, and I think it even has some bugs [...]

不好意思，没有呢，我玩Airsim也才一个多月，目前都是在仿真上在玩
Sorry, no, I've only been playing Airsim for more than a month. Now I'm playing on simulation.

machenxiang, 2019-07-22:
> 不好意思，没有呢，我玩Airsim也才一个多月，目前都是在仿真上在玩
> Sorry, no, I've only been playing Airsim for more than a month. Now I'm playing on simulation.
OK, thank you. Can we add a qq friend? 1243858461

saihv, 2020-01-20:
Closing due to inactivity.

closed 2020-01-20 by saihv