microsoft/AirSim #2058: Car PID throttle control instability

NmBoyd, 2019-07-03:
Hello,

I'm doing PID trajectory control using AirSims car model. I've been running into strange issues regarding PID control for the throttle of the vehicle. I've setup the PID such that the braking is equivalent to the negative of the throttle, provided there is any overshoot. For the most part this works extremely well, however for some reason I get instability in the PID controller as soon as I pass 25mph (~12m/s). Examining the torque curve I know the nonlinear relationship between the vehicle speed and throttle acceleration and my PID should be ale to compensate for it across almost all speeds.

Here is the general code for throttle controller:

```
throttleComp = 0.5*errorV;
dthrottleComp = 0.15*derrorV;
ithrottleComp = 0.025*ierrorV; 
 
controls.throttle = throttleComp + dthrottleComp + ithrottleComp;
			
if (controls.throttle < 0)
{
        controls.brake = -1*controls.throttle;
	controls.throttle = 0;
}
else
	controls.brake = 0;
```
You can see the response graphs in the imgur link.
https://imgur.com/a/DOqQ4JD

You'll see how the overall vehicle speed gets steady state oscillations within 10% of its desired setpoint but never settles even with the integral term. I double checked to make sure it wasn't a PID related issue by running it through AirSim with a gain optimizer a couple hundred times looking for values that cause no steady state oscillation but couldn't find anything.

My guess is that Airsim/Unreal PhysX has some state based modifier that changes the braking control values to something stronger at the 25mph zone, maybe to a "Bang-Bang" type of control, or maybe there is just a bug, but I'm not sure. Another possibility may be that the braking scales way differently than throttle and needs to be controlled independent of the throttle. Any help would be appreciated.

Hezzzzzy, 2019-07-05:
I also meet the similar problem ， I solved it by using he incremental PID and limiting the MAX throttle, because when your throttle = 1, you get the MAX power, there is no difference between 1000 throttle or 1 throttle,  when I done these, there is no shock in the velocity curve

NmBoyd, 2019-07-05:
I am capping the throttle and braking so it doesnt exceed 1, but it's more just for readability since it caps internally. I've made some headway and decreased it by splitting the controller into an independent braking and throttle controll (2 PID's with different gains based on their respective curves if they have one). However I still dont get 0 steady state error. 

Can you clarify what you mean by incremental PID  just to make sure we are on the same page? I believe I know what you are referring to and will try it tomorrow.

Hezzzzzy, 2019-07-05:
> I am capping the throttle and braking so it doesnt exceed 1, but it's more just for readability since it caps internally. I've made some headway and decreased it by splitting the controller into an [...]

I mean , PID control the increment of throttle ， not control the throttle directly, I am not sure it can solve your problem

jonyMarino, 2020-08-17:
@NmBoyd do you still have this issue?

closed 2020-08-17 by NmBoyd