microsoft/AirSim #2059: what is the size of the multirotor in unreal

vxgu86, 2019-07-04:
I made a cylinder with radius 1 m in top view, and I found that the multirotor is almost about 1 meter length, should it be 450 or so?
and I have been thinking about this for a long time, what is the metric standard of unreal by 1m ? did it have any relation with airsim?

vxgu86, 2019-07-04:
sorry to add one question
I found that the debug view has a position parameter, which  shows like this:
position (98.201) (90.211,78.225,-256)
is this the global position for the multirotor or  the local position?
and what is the first (98.201) means?

bwassim, 2019-09-04:
It also appears to me that the size of the multirotor is 1mx1m

madratman, 2020-08-11:
1) Open `Quadrotor1` mesh 
[image]

2) Change view to Top / Left, and enable "show bounds" (this will show a grid of 1cm along with the mesh bounds). 
The size is already mentioned in the top left (98x98x29 cm). 
[image]

3) Top view: (see the edges of the blue rectangle)
[image]
[image]

4) Left view: (see the edges of the blue rectangle)
[image]

closed 2020-08-11 by madratman