microsoft/AirSim #2061: Cannot see drone when playing

nabeelyousfi, 2019-07-08:
I followed Chris's AirSim from scratch video and followed each and every step. But when I hit play I don't see my drone. I just see a FPV or maybe it's just the camera.

[image]

I can move around using W, A, S, D/Arrow keys but I want to see the drone. I toggled B, F keys but nothing happens.

msb336, 2019-07-12:
Did you change the game mode to AirSimGameMode? Click the down arrow under `Blueprints`, navigate down to `GameMode`, and make sure `AirSimGameMode` is selected.

PatrickUtz, 2019-07-15:
Try changing (or adding) the "ViewMode" setting in the `settings.json` file to "". This is how I have it set up:
[image]

More info on this setting can be found here: https://github.com/microsoft/AirSim/blob/master/docs/settings.md

Note that if you are using Linux, the settings.json file is located under /home/<yourusername>/Documents/AirSim

saihv, 2019-10-30:
Closing due to inactivity. 

Please make sure the game mode is AirSimGameMode, and that your vehicle type is Multirotor or Car, not ComputerVision.

closed 2019-10-30 by saihv