microsoft/AirSim #2067: AirSim API

nalfano16, 2019-07-09:
I am looking to find out how to use APIs in my project however there are no clear answers on how to use them. A quick breakdown of my setup: I am using a steering column with a steering wheel to control a vehicle. I am using CAN data from the column as the steering input. The CAN side part is all working so I don't need a ton of help here. However, attaching that data to variables that can be used in unreal is the issue. I have a visual studios solution that runs the unreal game and then a visual studios solution that runs a server that connects to a tablet (used for various inputs), unreal, and the CAN messages. Where does the API need to go in order for it to connect to unreal, what are the functions I need to control the steering, throttle, brake, handbrake controls? What do I need to do to enable the connection? All of this information is extremely hard to find and since my case is a little different than most I am a little lost. Thanks for any help.

ayalakevin, 2019-07-11:
Hi nalfano,

To run the api's:
1. Have the visual studio and unreal engine running.
2. Make sure you have pressed play in the unreal engine.
3. Open Api and simply run it. The api should return a confirmation and run its code.
You can try the premade Api's and see if they run.

saihv, 2020-01-20:
Closing due to inactivity.

closed 2020-01-20 by saihv