microsoft/AirSim #2070: How to slow down a drone

ayalakevin, 2019-07-11:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------
Hi Everyone,

I have a simple setup where two drones are to move in an X direction to end up colliding with each other. I have calculated the distance between them and I want use that distance to avoid collision by slowing one down while the other passes and vice versa. However, I can’t get moveByVelocity to work in this case and when I call moveByVelocity, where do I identify that I want Drone1 to slow down or Drone2. Any help is greatly appreciated, thank you in advance.

mahieyin-rahmun, 2019-07-12:
https://github.com/microsoft/AirSim/blob/master/PythonClient/airsim/client.py#L284

if you look at the code you can see that there's a keyword arg called "vehicle_name". While calling the function you can set vehicle_name='Drone1' to control Drone1.

ayalakevin, 2019-07-12:
Thank you mahieyin-rahmun!

Didn't know the master existed, great help!

closed 2019-07-12 by ayalakevin