microsoft/AirSim #2071: multirotor have the same position

ldgcug, 2019-07-11:
I have set up multiple drones in settings.json

settings.json
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode":"Multirotor",
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 640,
        "Height": 480,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      }
    ]
  },

  "Vehicles":{
    "Drone1":{
      "VehicleType":"SimpleFlight",
      "X":4,"Y":0,"Z":0
    },
    "Drone2":{
      "VehicleType":"SimpleFlight",
      "X":8,"Y":0,"Z":0,"Yaw":180
    }
  }
}
```

But I got the same location by using the API
```

state1 = client.getMultirotorState(vehicle_name="Drone1")
s = pprint.pformat(state1)
print("state: %s" % s)
state2 = client.getMultirotorState(vehicle_name="Drone2")
s = pprint.pformat(state2)
print("state: %s" % s)
```

**logout:**
state: <instance> {   u'collision': <instance> {   u'has_collided': False,
    u'impact_point': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'normal': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'object_id': -1,
    u'object_name': u'',
    u'penetration_depth': 0.0,
    u'position': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'time_stamp': 0},
    u'gps_location': <instance> {   u'altitude': 123.34649658203125,
    u'latitude': 47.64151830548043,
    u'longitude': -122.14036499090417},
    u'kinematics_estimated': <instance> {   u'angular_acceleration': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'angular_velocity': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'linear_acceleration': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'linear_velocity': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'orientation': <instance> {   u'w_val': 1.0,
    u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'position': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': -0.14649589359760284}},
    u'landed_state': 0,
    u'rc_data': <instance> {   u'is_initialized': False,
    u'is_valid': True,
    u'left_z': -431602080.0,
    u'pitch': 431602080.0,
    u'right_z': -431602080.0,
    u'roll': -431602080.0,
    u'switches': 52672,
    u'throttle': -215801040.0,
    u'timestamp': 0,
    u'vendor_id': u'',
    u'yaw': -431602080.0},
    u'timestamp': 1562838174842467328}
state: <instance> {   u'collision': <instance> {   u'has_collided': False,
    u'impact_point': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'normal': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'object_id': -1,
    u'object_name': u'',
    u'penetration_depth': 0.0,
    u'position': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'time_stamp': 0},
    u'gps_location': <instance> {   u'altitude': 123.34649658203125,
    u'latitude': 47.64155423809056,
    u'longitude': -122.14036499104174},
    u'kinematics_estimated': <instance> {   u'angular_acceleration': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'angular_velocity': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'linear_acceleration': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'linear_velocity': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': 0.0},
    u'orientation': <instance> {   u'w_val': 3.1391647326017846e-07,
    u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': -1.0},
    u'position': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': -0.14649589359760284}},
    u'landed_state': 0,
    u'rc_data': <instance> {   u'is_initialized': False,
    u'is_valid': True,
    u'left_z': -431602080.0,
    u'pitch': 431602080.0,
    u'right_z': -431602080.0,
    u'roll': -431602080.0,
    u'switches': 52672,
    u'throttle': -215801040.0,
    u'timestamp': 0,
    u'vendor_id': u'',
    u'yaw': -431602080.0},
    u'timestamp': 1562838174842467328}

Why?

ayalakevin, 2019-07-11:
hi Idgcug,

Every drone will act as their own origin point. Although in settings they are separated by 4 in the x direction, in unreal they see themselves as  0,0,0 in their own frame. Therefore, you need to choose one drone as the main frame and the other will act off that one. One way I did it was to find their specific estimated kinematic positions, x,y,z, and have one drone x+#,y,z so that the second drone will math the difference in settings and API.

ldgcug, 2019-07-12:
> hi Idgcug,
>
> Every drone will act as their own origin point. Although in settings they are separated by 4 in the x direction, in unreal they see themselves as 0,0,0 in their own frame. Therefore, you [...]

Thank you, you are right, no matter what the coordinates of the drone I set in settings.json, the coordinates I get will be (0,0,0), so I can give an initial difference value ( x, y, z) to get the real coordinates, as you said

closed 2019-07-12 by msb336