microsoft/AirSim #2074: Change of coordinates in Landscape environment

Baldins, 2019-07-14:
Hi, 

When I want to fly the drone in the Mountain Landspace environment, the drone starts randomly from different points.
 Is there a way to fix the starting origin to a specific point? 

Additionally, every time I start the project I noted that the coordinate of my desired origin changes. My guess is that everything is relative to the actual position of the drone. 
Is there a way to move to a specific point in absolute coordinate? 

I tried to save the absolute coordinate of the drone and change the origin in the setting, but I keep having the same problem over and over. 

If someone can help me with it, it would be really appreciated! 

Thanks!

PatrickUtz, 2019-07-15:
I had the same issue and figured out that AirSim was using the coordinates set in my `settings.json` file. I would recommend not using the Unreal Editor coordinates pane and instead modify the `"X": 4, "Y": 0, "Z": -9` section of `settings.json` as shown here:
[image]

Note that if you are using Linux, the `settings.json` file is located under `/home/<yourusername>/Documents/AirSim`

Baldins, 2019-07-15:
Thank you for replying! 
I tried what you suggested but I keep having the same problem. Every time I restart, the initial position of the drone changes. Do you have any idea of why is this happening? 
Thank you very much for your help!

Baldins, 2019-07-15:
Never mind! I needed to restart! It is starting from the same point now ! Thank you very much for the tip!!!

closed 2019-07-15 by Baldins