microsoft/AirSim #2077: How to Set Segmentation ID For PlayerStart Components

PatrickUtz, 2019-07-15:
I am wondering if it is possible to set the segmentation ID for the PlayerStart component(s). According to the [camera api docs](https://microsoft.github.io/AirSim/docs/image_apis/#camera-apis), the `simSetSegmentationObjectID()` function works for `UStaticMeshComponent` or `ALandscapeProxy` meshes. As seen below, I am able to segment (using the infrared setting) regular _static_ meshes like so:
[image]
[image]

Yet I am unable to do so for PlayerStart_2 (Drone2 in my case) as seen below:
[image]
[image]

Any form of help would be greatly appreciated. Thank you.

machampagne1, 2019-07-29:
Hello !
I'm currently looking into giving the object a segmention from python (I can't atm) but if you want to give a blueprint (say the drone) it's own segmentation, you can add these 2 nodes somewhere in the event graph or in the construction script : 
[image]

You need to _get_ the static mesh and use this as the target. the value will determine the color of the segmentation (just make sure it's different than from your floor)

PatrickUtz, 2019-07-30:
Fantastic, will give this a shot. Thank you!

msb336, 2019-09-25:
I do not think `PlayerStart` objects have any visible properties. If you are attempting to view drones, they are defined as  `BP_FlyingPawn{instance}`

closed 2019-09-25 by msb336