microsoft/AirSim #2078: What is the origin of GPS data

Hezzzzzy, 2019-07-16:
hi guys,
does anyone knows , what is the origin of GPS data which get from getGpsData()， 
thx a lot

Hezzzzzy, 2019-07-16:
@msb336 beg for helping! thx

Hezzzzzy, 2019-07-16:
I have found it in AirsimSettings.hpp, thx

Nianan0505, 2019-11-14:
hi，would you help me how to get GPS data about car.
I cannot find or use api

Hezzzzzy, 2019-11-23:
> hi，would you help me how to get GPS data about car.
> I cannot find or use api

你要先在车上安装了GPS的sensor才能直接获取GPS，而且我不推荐你这样做，因为之前做数据校验的时候，发现airsim的很多数据存在问题，建议直接获取UU坐标系的RT，自己转GPS，这样比较灵活也能保证精确度

closed 2019-11-23 by Hezzzzzy

Nianan0505, 2019-12-03:
> > hi，would you help me how to get GPS data about car.
> > I cannot find or use api
> 
> 你要先在车上安装了GPS的sensor才能直接获取GPS，而且我不推荐你这样做，因为之前做数据校验的时候，发现airsim的很多数据存在问题，建议直接获取UU坐标系的RT，自己转GPS，这样比较灵活也能保证精确度

非常感谢您的解答，我还有不明白的地方：
1.在车上安装gps是否等同于我在settings.json中设置如下：
 "Gps": {  "SensorType": 3, "Enabled" : true },
2.如何直接获取UU坐标系的RT呢？
再次表示感谢！！！

Hezzzzzy, 2019-12-03:
1.是这样设置，但是具体需要设置哪些参数有点忘了，有一段时间没用了，但是我记得指导文档上有示例。
2.你需要去读一下airsim api c++版本的代码，我只能给你大概说明位置，airsim获取imu流程是 通过UE4自身的api获取UU坐标系的RT（UU坐标系是UE4内部世界定义的坐标系）——> 坐标系对齐 ——> 做一些计算转换，比如起始点什么的，最终转为airsim的IMU输出，我的做法是，airsim源码里有一个地方写了一个基础函数，即是获取UU坐标系的RT，然后做转换，我在这里修改了这个函数，把RT抓出来了，这样做比较方便的地方是这个函数是根据UE4同步刷新的，你不需要重新写回调函数，你去找找这个函数吧，我忘了叫什么名字了。

Nianan0505, 2019-12-03:
> > > > hi，would you help me how to get GPS data about car.
> > > > I cannot find or use api
> > > 
> > > 
> > > 你要先在车上安装了GPS的sensor才能直接获取GPS，而且我不推荐你这样做，因为之前做数据校验的时候，发现airsim的很多数据存在问题，建议直接获取UU坐标系的RT，自己转GPS，这样比较灵活也能保证精确度
> > 
> > 
> > 非常感谢您的解答，我还有不明白的地方：
> > 1.在车上安装gps是否等同于我在settings.json中设置如下：
> > "Gps": { "SensorType": 3, "Enabled" : true },
> > 2.如何直接获取UU坐标系的RT呢？
> > 再次表示感谢！！！
> 
> 1.是这样设置，但是具体需要设置哪些参数有点忘了，有一段时间没用了，但是我记得指导文档上有示例。
> 2.你需要去读一下airsim api c++版本的代码，我只能给你大概说明位置，airsim获取imu流程是 通过UE4自身的api获取UU坐标系的RT（UU坐标系是UE4内部世界定义的坐标系）——> 坐标系对齐 ——> 做一些计算转换，比如起始点什么的，最终转为airsim的IMU输出，我的做法是，airsim源码里有一个地方写了一个基础函数，即是获取UU坐标系的RT，然后做转换，我在这里修改了这个函数，把RT抓出来了，这样做比较方便的地方是这个函数是根据UE4同步刷新的，你不需要重新写回调函数，你去找找这个函数吧，我忘了叫什么名字了。

好的，真的非常感谢