microsoft/AirSim #2084: Multi HITL Airsim with px4

Dan-Moss, 2019-07-17:
PX4 version: 1.9.2
Using the Simple maze binary in UE4

I can have 2 single instances of the px4 running and can fly them both individually when only one is connected via serial at a time. But if i connect them both at the same time via usb i get:

connecting to px4 over serial port: COM8, baude 115200 ...
Connected to px4 over serial
connecting to px4 over serial port: COM7, baude 115200 ...
not receiving any messages from HIL

setting file:
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "PX2_1": {
      "VehicleType": "PX4Multirotor",
      "SerialBaudRate": 115200,
      "SerialPort": "COM7",
      "UseSerial": true,
      "SimCompID": 1,
      "SimSysID": 11
    },
    "PX2_2": {
      "VehicleType": "PX4Multirotor",
      "SerialBaudRate": 115200,
      "SerialPort": "COM8",
      "UseSerial": true,
      "SimCompID": 2,
      "SimSysID": 12,
      "X": -4,
      "Y": 0,
      "Z": 0
    }
  }
}

Is there any setting or configs i am missing or not setting?

Dan-Moss, 2019-08-12:
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
"Vehicles": {
    
    "PX401": {
      "VehicleType": "PX4Multirotor",
      "VehicleSysID": 10,
      "SerialPort":  "COM7"
    },
    "PX402": {
      "VehicleType": "PX4Multirotor",
      "VehicleSysID": 20,
      "SerialPort": "COM8",
      "X": -4,
      "Y": 0,
      "Z": 0
    }
  },
  "SubWindows": [
    {"WindowID": 0, "ImageType": 0, "CameraName": "1", "Visible": true},
    {"WindowID": 1, "ImageType": 0, "CameraName": "2", "Visible": true},
    {"WindowID": 2, "ImageType": 0, "CameraName": "3", "Visible": true}
  ]
}

These parameters work, in 1.8.1, ensure that the SYS_ID is set to the same com port in both the FCU Params and settings file.

madratman, 2020-08-11:
similar issue https://github.com/microsoft/AirSim/issues/1599.

closed 2020-08-11 by madratman