microsoft/AirSim #2085: Set personalized reset() and landing Issue

Baldins, 2019-07-17:
Hi all, 

I have the following 2 problems that I am not able to solve:

- I would like to reset the drone on a specific point in the airsim environment. However, the coordinates in the setting.json are different from the one in the simulated environment and even if I use client.moveToPositionAsync() is not going where I am asking for. 

- I would like to land on a specific point but the drone keeps hovering. Is there any way I can solve it? When I use client.landAsync().join() the drone doesn't actually land. 

Please, let me know if someone can help me with these issues! 

Thank you very much!!!

Francesca

AloshkaD, 2019-07-17:
No matter where you start your drone in the config file, the API will return a x=0, y=0 coordinates because it considers that location as the origin point. Hence, it will always reset to those coordinates in the sim. 
As to moving the dron, I use something like this 
``` def movement(speed_x, speed_y, speed_z, duration,vehicle_name=''):
        
        pitch, roll, yaw  = self.getPitchRollYaw(vehicle_name)
        vel = self.simGetGroundTruthKinematics(vehicle_name).linear_velocity
        drivetrain = DrivetrainType.ForwardOnly
        yaw_mode = YawMode(is_rate= False, yaw_or_rate = 0)

        moveByVelocityAsync(vx = vel.x_val + speed_x,
                            vy = vel.y_val + speed_y,
                            vz = vel.z_val + speed_z,
                            duration = duration,
                            drivetrain = drivetrain,
                            yaw_mode = yaw_mode,vehicle_name= vehicle_name)
        
``` 
for landing I use ```moveToZAsync(0, 1,vehicle_name)```
and call it like this 

```movement(0.5, 0, 0, duration,vehicle_name)``` for example

AloshkaD, 2019-07-17:
also, see this https://github.com/microsoft/AirSim/issues/2055#issuecomment-511553197

Baldins, 2019-07-18:
Thank you for your reply! 

The problem with moveByVelocityAsync() is that I can not assign a specific point and I am trying to land the center of a platform with high precision. 

Do you have any idea of what I can do to make it happens? 

Basically, what I am doing is driving the drone manually to the platform (with a remote controller) and printing the kinematics so that I can know my original coordinates. Then I launch my API code to move the drone to a random point around the platform and finally coming back and landing on my original configuration. 

Please, let me know if you can help me with it! 

Thank you very much!!!

AloshkaD, 2019-07-18:
Your approach is correct. 
1- You need to fly the drone and get the coordinates and then convert them, so you can use them in the setting file. Another approach is to set the setting with coordinates and keep testing it until you are exactly over the coordinates that you need. 
2- for moving the drone to the desired trget, I simply did
```if  0  < self.episodeN < 10 :
            print('333333333333333333333 moving level 1 3333333333333333333333333')
            target1_new_position=[6,-1,self.z]

            trackgym.moveToPositionAsync(target1_new_position[0],target1_new_position[1],target1_new_position[2], 5, vehicle_name="Target1").join()
            target1_position = trackgym.simGetGroundTruthKinematics(self.target1_vehicle_name).position```
you can find it in line 203 in my code[https://github.com/AloshkaD/TrackGym-baseline/blob/master/gym_trackairsim/envs/TrackGym.py](url)

3- Obviously, after moving the drone to the position using the code above, you can test the landing command that I provided above. I hadn't tested it so you have to go ahead and test it.

AloshkaD, 2019-07-18:
you can find it in line 203 in my code[https://github.com/AloshkaD/TrackGym-baseline/blob/master/gym_trackairsim/envs/TrackGym.py](url)

3- Obviously, after moving the drone to the position using the code above, you can test the landing command that I provided earlier. I hadn't tested it so you have to go ahead and test it.

Baldins, 2019-07-18:
It seems like it is working now!!! 
Thank you very much for the tips!!! :D

AloshkaD, 2019-07-18:
You are welcome, man. Glad it worked :)

closed 2019-07-18 by Baldins