microsoft/AirSim #2087: Collect Camera and IMU data from API

Baldins, 2019-07-18:
Hi all, 

I would like to collect images and IMU data in a single code while the drone is flying. 

So far, I am launching 3 different codes, one for controlling the drone and separately a code to get IMU data and another one for collecting Images. 
Is there any way I can just write a single code for it?

The main problem for me was that the collection of images or IMU data stops while the drone moves and restarts after the drone finish moving. 
Is there any way I can keep collecting data while the drone moves from the same python code? 

Thank you in advance to whom will be able to help me! :)

AloshkaD, 2019-07-18:
You need to collect what you need to collect in a loop, while moving without .join(). are you doing that?

Baldins, 2019-07-18:
No. I am not doing it cause in the loop I want to reach 3 different points and if I do not use .join() it goes directly to the last point. 

Additionally, if I want to collect the IMU data, it waits first for the image to be collected and then start the IMU collection and viceversa, while I aim to do it at the same time.

For now, I avoided this issue just launching different codes but It would be nice if I can do everything at once.

closed 2019-08-22 by msb336