microsoft/AirSim #2089: 0 Byte Images

Baldins, 2019-07-19:
Hi, 

while collecting images I get some of them of size 0 B.
Does someone know why it happens and what can I do for not let this happening anymore?

Thanks in advance!!!

AloshkaD, 2019-07-20:
This is a common problem, especially with drones. Add a sanity check to the image acquisition method that will generate a black image if your pipeline requires a continuous image stream, and you will lose a one or few frames but your pipeline won't break. Here's a sample code 

```
responses = self.simGetImages([ImageRequest(0, ImageType.DepthPerspective, True, False)],vehicle_name)
        if (responses[0] == None):
            print("Camera is not returning image, please check airsim for error messages")
            sys.exit(0)
        img1d = np.array(responses[0].image_data_float, dtype=np.float)
        img1d = 255/np.maximum(np.ones(img1d.size), img1d)
        if (len(img1d) != (144*256)):
            print('The depth camera returned bad data so Im creating zero array to not break the training')
            img1d=np.ones(144*256)
            img2d = np.reshape(img1d,(144, 256))```

AloshkaD, 2019-07-20:
And I apologize about referring to you as a "man" earlier. I didn't realize that you are a mam. I tend to be ignorant sometimes with all the workload in my head.

Baldins, 2019-07-25:
Thank you! This is great! :)

closed 2019-07-25 by Baldins