microsoft/AirSim #2090: Cannot connect to the Airsim while training without error message

Hoshea7, 2019-07-19:
Hello,

I am trying to use the DQN algorithm to train the agent in Airsim, however, sometimes when I trained it after a certain time episode, the connections between the Python client and Airsim is strangely not working and also without any error messages popped out. 

Here below is the image that I do the screen cut about this question. The specific time that I do the screen cut of it is after I lose the connections between training process, and then I tried to reconnect the Airsim after reopen the Pycharm and Airsim simulator. Then, after waiting for a long time which is longer than the normal connected time, it still gets no connections and no messages to show if connected or not.
[image]

Could anyone tell me what is this issue? and if possible how to fix it ?

msb336, 2019-09-26:
I believe [this pr](https://github.com/microsoft/AirSim/pull/2146) will fix your problem. Please try updating to current master and rebuilding.

saihv, 2020-01-20:
Closing due to inactivity.

closed 2020-01-20 by saihv