microsoft/AirSim #2091: Very start error with build.cmd

guiwecgdiu, 2019-07-19:
I am tried to download the newest release, but I can't run the build.cmd with below error.
I used an administor acces to run the native command prompt in visual 2019. 
Wether I must use visual 2017 to generate the plugin or the error is for other reason?
Error here:
########################################
D:\AirSim\AirSim_A\AirSim-master\AirSim-master>build.cmd
Found cmake version: 3.14.19050301-MSVC_2

*****************************************************************************************
Downloading rpclib
*****************************************************************************************
Access is denied.
Access is denied.
Could Not Find D:\AirSim\AirSim_A\AirSim-master\AirSim-master\external\rpclib.zip
Unable to download high-polycount SUV. Your AirSim build will use the default vehicle.

#### Build failed - see messages above.

D:\AirSim\AirSim_A\AirSim-master\AirSim-master>
################################################

saihv, 2019-10-30:
Currently we only support VS 2017. Please try to run the install script in the VS 2017 Developer command prompt with admin privileges.

saihv, 2020-01-20:
Closing due to inactivity.

closed 2020-01-20 by saihv