microsoft/AirSim #2092: How to transformed the Local Ned coord to global Ned coord

hkflowcl, 2019-07-19:
I want to run some trajectories and then do the alignment. Because Local Ned coord was used so each case use (0,0,0) as origin . I want to set an anchor Point like (0,0,0) so that I may get the world Pose of each startPoint. According to some page, I try to set origin-geopoint as(0,0,0) and add the global offset. but the startPoint still is (0,0,0) even I start at different Point.  
I will appreciate a lot if someone can help to clarify the question.

AloshkaD, 2019-07-20:
Typically, when you set coordinates in the settings.json e.g., (2,4,-2) the drone/car will start from that point in unreal, but if you call the position from the api it will give you (0,0,1.8) or something. In order to correct for this I introduce the coordinated in the settings.json as an offset  and add it to reported position from the api, like this
```class dotdict(dict):
    """dot.notation access to dictionary attributes"""
    __getattr__ = dict.get
    __setattr__ = dict.__setitem__
    __delattr__ = dict.__delitem__
offset = {'x_val':2.0,'y_val':0.0,'z_val':-2.0}
offset = dotdict(offset)

        self.init_goal = airgym.simGetGroundTruthKinematics(self.target1_vehicle_name).position 	#[137.5, -48.7,-4]
        self.goal={'x_val':self.init_goal.x_val+offset.x_val, 'y_val':self.init_goal.y_val+offset.y_val,'z_val':self.init_goal.z_val}
        self.goal = dotdict(self.goal) 
         
        self.distance = np.sqrt(np.power((self.goal.x_val),2) + np.power((self.goal.y_val),2) + np.power((self.goal.z_val),2))
```

hkflowcl, 2019-08-05:
> Typically, when you set coordinates in the settings.json e.g., (2,4,-2) the drone/car will start from that point in unreal, but if you call the position from the api it will give you (0,0,1.8) or something like that. In order to correct for this I introduce the coordinates in the settings.json as an offset and add it to reported position from the api, like this
> 
> ```
> class dotdict(dict):
>     """dot.notation access to dictionary attributes"""
>     __getattr__ = dict.get
>     __setattr__ = dict.__setitem__
>     __delattr__ = dict.__delitem__
> offset = {'x_val':2.0,'y_val':0.0,'z_val':-2.0}
> offset = dotdict(offset)
> 
> self.init_goal = simGetGroundTruthKinematics(self.vehicle_name).position 	#[137.5, -48.7,-4]
> self.goal={'x_val':self.init_goal.x_val+offset.x_val, 'y_val':self.init_goal.y_val+offset.y_val,'z_val':self.init_goal.z_val}
> self.goal = dotdict(self.goal) 
> self.distance = np.sqrt(np.power((self.goal.x_val),2) + np.power((self.goal.y_val),2) + np.power((self.goal.z_val),2))
> ```

Thanks a lot

closed 2019-08-05 by hkflowcl