microsoft/AirSim #2096: image sent to

Nianan0505, 2019-07-24:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

machenxiang, 2019-07-30:
Which ros do you use? rospy or roscpp?

Nianan0505, 2019-07-31:
I use roscpp

Do you meet the same problem？

machenxiang, 2019-07-31:
> I use roscpp Do you meet the same problem？
> […](#)

 我用rospy可以拿到图像。roscpp正在做。
I can get the image via rospy. Now i am using roscpp.

Nianan0505, 2019-07-31:
我用rospy也出现了同样问题，airsim_ros_node不是自己写的，用的官方提供的。
我的邮箱是3150100638@zju.edu.cn 我有一些问题想请教你，可以进一步交流吗
I meet the same question via rospy. I am using airsim_ros_node provided by airsim.
May we have further communication? Here is my email.

Nianan0505, 2019-07-31:
> > I use roscpp Do you meet the same problem？
> > […](#)
> 
> 我用rospy可以拿到图像。roscpp正在做。airsim_ros_node是你自己的写的cpp吗？
> I can get the image via rospy. Now i am using roscpp. airsim_ros_node is written by yourself?

我用rospy遇到了同样的问题，用的官方提供的airsim_ros_node
我的邮箱是3150100638@zju.edu.cn  我们可以进一步交流吗
I meet the same question via rospy. I use airsim_ros_node provided by airsim
May we have further communication, please? Here is my email.