microsoft/AirSim #2099: when i run ros as airsim_ros_pkgs roslaunch airsim_ros_pkgs airsim_node.launch it occur error!

machenxiang, 2019-07-25:
error
Exception raised by the API:
rpclib: server could not find function 'getImuData' with argument count 2.

Nianan0505, 2019-07-31:
我前几天也遇到了这个问题，你现在解决了吗？
I met the same question several days ago. 
Do you solve it now?

machenxiang, 2019-08-06:
> 我前几天也遇到了这个问题，你现在解决了吗？
> I met the same question several days ago.
> Do you solve it now?
没有，我换了一台电脑然后又出现了新的错误。你解决了吗？
No，I change another computer for simulation,then it occurs other errors.
error
Exception raised by the API:
rpclib: function 'getMultirotorState' (called with 1 arg(s)) threw an exception. The exception is not derived from std::exception. No further information available.
Did you solve the problem?

machenxiang, 2019-08-06:
> 我前几天也遇到了这个问题，你现在解决了吗？
> I met the same question several days ago.
> Do you solve it now?
我们建了一个qq群，有兴趣可以加以下只不过人很少不算你3个人？
We create a qq group. If you are interested in it,you can join us. But now it's only 3 people.

Nianan0505, 2019-08-06:
> > 我前几天也遇到了这个问题，你现在解决了吗？
> > I met the same question several days ago.
> > Do you solve it now?
> > 我们建了一个qq群，有兴趣可以加以下只不过人很少不算你3个人？qq群号：644726559
> > We create a qq group. If you are interested in it,you can join us. But now it's only 3 people. qq  group ID: 644726559

I am really glad to join you. But is the ID number correct? I am sorry， but I cannot search and find the group.

machenxiang, 2019-08-06:
> I am really glad to join you. But is the ID number correct? I am sorry， but I cannot search and find the group.

你先加我qq吧 1243858461

mbheir, 2019-09-30:
I have the same issue, did anyone figure it out?

chjz1024, 2019-11-17:
@mbheir @machenxiang @Nianan0505 
You need to change the settings.json as specified in https://github.com/microsoft/AirSim/tree/master/ros/src/airsim_tutorial_pkgs/settings then everything will continue to work

heleidsn, 2019-12-05:
> @mbheir @machenxiang @Nianan0505
> You need to change the settings.json as specified in https://github.com/microsoft/AirSim/tree/master/ros/src/airsim_tutorial_pkgs/settings then everything will continue to work

I use the same settings.json as you said, but the problem still exists. Moreover, I cannot get images in rviz. Did anyone run this correctly with ROS?

Nianan0505, 2019-12-05:
> I use the same settings.json as you said, but the problem still exists. Moreover, I cannot get images in rviz. Did anyone run this correctly with ROS?

I do not recommend you to use ‘getImuData’ function.
You can see this issue [https://github.com/microsoft/AirSim/issues/1909](url) to use the correct way to get IMU data.
What's your problem about images?

heleidsn, 2019-12-05:
Hi @Nianan0505, 

Thanks for your reply! 
When I want to see image of the front camera in rviz, there is a problem:
`[ERROR] [1575559110.136588745]: Error loading image: OGRE EXCEPTION(2:InvalidParametersException): Stream size does not match calculated image size in Image::loadRawData at /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreImage.cpp (line 283)
`
Maybe I just need to check the OgreImage.cpp or change the image size...

mbheir, 2019-12-05:
@heleidsn I have the same issue. It seems when the error occur, ROS isn't getting any data from Airsim. For me the error occurs when I'm running any of the unreal maps downloaded from Airsim. However, when I run Blocks, or build my own projects, Airsim seems to run just fine with ROS. 

The strange thing is, I'm using the same settings.json file as several others, and they don't get the error at all. I've also done a lot of rebuilds, without any results.

heleidsn, 2019-12-05:
@mbheir @Nianan0505 Yes, the error occurs when running with downloaded maps from the latest Linux release. When I open the environment with UE4editor, it works fine and I can get the image correctly. I have no idea why it doesn't work with the downloaded environment...

xuhao1, 2020-01-04:
@heleidsn After hours of debugging I found that the download environment gives RGBA image instead of RGB, so image capture will failed. I have no idea about the IMU, I will turn to self build instead

heleidsn, 2020-01-04:
@xuhao1 Thank you very much! BTW, I like your profile photo~

closed 2021-08-19 by jonyMarino