microsoft/AirSim #2100: Get Images  while moving from API

Baldins, 2019-07-25:
Hi, 

Is it possible to collect Images while moving with the drone from API? 
Every time I see codes where the drone hovers and then collect images.

Thank you in advance!

msb336, 2019-07-25:
yes, `simGetImages` works while the drone is moving.

Baldins, 2019-07-25:
No, I want to constantly collect images while the drone is moving around.
Using that function I just collect a single image (or multiple images if used in a loop), but always after the drone finishes his task.
If I don't use the join() I can collect the images but then the drone doesn't finish its task.

msb336, 2019-07-25:
Unfortunately there is no image streaming API in airsim. To constantly collect images, I would recommend creating two airsim clients on two threads. One which handles motion commands, and another which handles collecting camera imagery in a loop.

Baldins, 2019-07-25:
I am already doing it, but I want to change the drone state estimation with the one I get by running my own algorithm and then update the controller with my current pose estimate. 
So I need to collect images and IMU data while moving in the same code. 
Is it not possible then?

msb336, 2019-07-25:
If you are operating in a control loop, data collection and control commands do not need to be asynchronous:
```
while not EndCondition:
    # collect sensor data
    sensor_data = client.getWhatever()
    # discern control input
    vx,vy,vz = getControl(sensor_data)
    # update velocity
    client.moveByVelocityAsync(vx,vy,vz, duration)
```

Baldins, 2019-07-25:
ya, in effect this should work - Thanks!

closed 2019-07-25 by Baldins

DavidKalajdzic, 2025-05-18:
One trick can be to pause and then resume the simulation while taking the picture, the join can be kept :
```
client.simPause(True)
images = client.simGetImages(...)
client.simPause(False)
client.moveByVelocityAsync(vx,vy,vz, duration).join()
```