microsoft/AirSim #2101: two Error when run unity demo

Amin-Tgz, 2019-07-25:
[image]

first one:

> NullReferenceException: Object reference not set to an instance of an object
AirSimUnity.Vehicle.OnApplicationQuit () (at Assets/AirSimAssets/Scripts/Vehicles/Vehicle.cs:162)

and second one:

> DllNotFoundException: AirsimWrapper
AirSimUnity.VehicleCompanion.InitDelegators () (at Assets/AirSimAssets/Scripts/Vehicles/VehicleCompanion.cs:85)
AirSimUnity.VehicleCompanion..cctor () (at Assets/AirSimAssets/Scripts/Vehicles/VehicleCompanion.cs:30)
Rethrow as TypeInitializationException: The type initializer for 'AirSimUnity.VehicleCompanion' threw an exception.
AirSimUnity.Vehicle.Start () (at Assets/AirSimAssets/Scripts/Vehicles/Vehicle.cs:58)
AirSimUnity.Car.Start () (at Assets/AirSimAssets/Scripts/Vehicles/Car/Car.cs:25)

car and drone dont move with arrow key or WASD

thanks

msb336, 2019-08-01:
This looks like the demo cannot find the AirLibWrapper shared library. Did you follow the [unity build instructions](https://github.com/microsoft/AirSim/tree/master/Unity)?

Amin-Tgz, 2019-08-03:
> 
> 
> This looks like the demo cannot find the AirLibWrapper shared library. Did you follow the [unity build instructions](https://github.com/microsoft/AirSim/tree/master/Unity)?

Yes, My OS is Windows (x64),
and all instruction has done successfully.
Is there another way to check install AirlibWrapper correctly?

closed 2019-08-28 by Amin-Tgz

shijianjian, 2022-05-26:
@Amin-Tgz can you post your solution here?