microsoft/AirSim #2103: [Reinforcement Learning] Center lane and frame frequency

MoritzGa, 2019-07-26:
Hello,

I am currently working on some reinforcement learning using AirSim and Unreal Engine (4.22) on Windows 10. I have two main questions:

- Is it possible to get the center lane from a custom environment such that one can compute the distance to it? Right now I've driven myself and recorded the center lane myself which is a bit suboptimal.
- How do I know at what frame frequency AirSim is running since I want to go through my main loop once every frame or step AirSim is running.

Thanks in advance!

rajat2004, 2019-07-26:
Hi,
AirSim's Physics Engine is running at 333 Hz (3000 ms) by default
https://github.com/microsoft/AirSim/blob/2450acfea564da1e426a5a5152ce4881ae76ccbb/Unreal/Plugins/AirSim/Source/SimMode/SimModeWorldBase.h#L65

You might want to look into using the Pause API for this
https://github.com/microsoft/AirSim/blob/master/docs/apis.md#pause-and-continue-apis

MoritzGa, 2019-09-09:
@rajat2004 Thanks for the reply.

closed 2019-09-09 by MoritzGa