microsoft/AirSim #2105: how can I get the depth scene when I transfer the simulation to a real drone?

vxgu86, 2019-07-29:
how can I get the depth scene when I transfer the simulation to a real drone?
should I get the perfect similar depth vision for the model to run the same function?

AloshkaD, 2019-07-29:
You can use many off-the-shelf depth sensors like Intel real-sense.
If you are asking whether a trained model using the sim's depth images will work using the real sensor, then the answer is it depends. You don't need perfectly same sensor, but something similar. If you have trained a model, you should have implemented some sort of augmentation that can generalize well.

vxgu86, 2019-07-30:
yep, I tried one with zed without working at all, just outputting random action, maybe it's because of the depth view?
I am testing by recording pics from ue4

AloshkaD, 2019-07-30:
I don't understand what you mean with "outputting random actions" and I won't be able to tell if it's because of the depth view because I have no clue what the application is. Try to print out the images and compare them to the simulation.

machenxiang, 2019-07-31:
> I don't understand what you mean with "outputting random actions" and I won't be able to tell if it's because of the depth view because I have no clue what the application is. Try to print out the images and compare them to the simulation.

 I'm sorry to bother you? Can I directly use AilLIb to control real drone(pixhawk)?  if it can work, how to do it?

AloshkaD, 2019-07-31:
Go to ```https://github.com/microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/docs/px4_setup.md```

There you will see a list of supported hardware

```## Supported Hardware

The following Pixhawk hardware has been tested with AirSim:

1. [3DR Pixhawk v2](https://3dr.com/support/pixhawk/)
2. [3DR Pixhawk mini](https://store.3dr.com/products/3dr-pixhawk)
```

vxgu86, 2019-07-31:
which version of the px4 and the hardware are you using?
the default latest version of px4_fmu-v2_default.px4 is not working with my pixhawk 2.4.8

> 
> 
> Go to `https://github.com/microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/docs/px4_setup.md`
> 
> There you will see a list of supported hardware
> 
> ```
> 
> The following Pixhawk hardware has been tested with AirSim:
> 
> 1. [3DR Pixhawk v2](https://3dr.com/support/pixhawk/)
> 2. [3DR Pixhawk mini](https://store.3dr.com/products/3dr-pixhawk)
> ```

closed 2020-09-04 by jonyMarino