microsoft/AirSim #2109: Increase mulitrotor speed for MovetoPosition in Unreal Engine

holdenKjerland-Nicoletti, 2019-07-30:
I am using the simpleflight drone with Unreal 4.18, and am trying to travel across a very large map. However, if I increase the speed for the MoveToPosition API past 12, I get really bad instability for the drone in the vertical direction. I assume this is because it is switching between moving forward and upward to go faster while not losing altitude, so it seems like I just need to change a setting to give the drone stronger motors.

Is there anything I can do to increase the max velocity of the drone for MoveToPosition API in Unreal?

cloudlakecho, 2020-07-01:
@holdenKjerland-Nicoletti Are you controlling the drone by the keyboard or computer code? If you use code, please share section of setting speed and moving the drone.

jonyMarino, 2020-11-30:
Closed due to incompleteness.
Please feel free to open a new issue if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing an issue
*Attach a screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-11-30 by jonyMarino