microsoft/AirSim #2113: Question about DQNDrone.py

rmdwjdrhkrkatk, 2019-08-03:
The example code DQNDrone.py has the following lines :
responses = client.simGetImages([ImageRequest(3, AirSimImageType.DepthPerspective, True, False)])
current_state = transform_input(responses)

When those lines are erased, the code does not run properly.
Awkward thing is that image does not used for the given algorithm. 

Why those lines are necessary for the example code?
If it is possible, I would like to get rid of those lines because those functions takes too much time.

AloshkaD, 2019-08-03:
This is an reinforcement learning algorithm and the acquired images are passed as states/observations to the DQN using the environment.
Hence you have 
```action = agent.act(current_state)```

and 
```agent.observe(current_state, action, reward, done)```

I highly recomend that you read more about RL environments and openai gym. even if the latter isn't used here, gym has great resources that will help you here.

rmdwjdrhkrkatk, 2019-08-05:
Oh, now I realize my mistake.
Thank you for your explanation!

AloshkaD, 2019-08-05:
U r welcome. Please don't forget to close the issue.

closed 2019-08-07 by msb336