microsoft/AirSim #2114: moveByVelocityAsync did not stop even after the duriation,should stop when another command come ?

vxgu86, 2019-08-04:
moveByVelocityAsync did not stop even after the duriation,should stop when another command come ?
the code below make the drone just move along x with no stop

moveByVelocityAsync(vx = 0.2 ,
                      vy = 0 ,
                      vz = 0,
                      duration = duration + 1,
                      drivetrain = drivetrain,
                      yaw_mode = yaw_mode).join()

msb336, 2019-08-07:
This is a known bug. Unless you are traveling towards the ground, there are no stop commands after the duration of moveByVelocityAsync() is complete. To stop your drone, you will need to do something like the following:

```
moveByVelocityAsync(vx,vy,vz,duration).join()
moveByVelocityAsync(0,0,0,1)
```

vxgu86, 2019-08-10:
got it,thanks.

closed 2019-08-10 by vxgu86

Rutsow, 2020-08-05:
> This is a known bug. Unless you are traveling towards the ground, there are no stop commands after the duration of moveByVelocityAsync() is complete. To stop your drone, you will need to do [...]

I test the solution, and I then call the getvelocity(), I find that the velocity is still [vx, vy, vz] not [0,0,0]. Is taht right?