microsoft/AirSim #2117: NameError: name 'DistanceSensorData' is not defined

rajatthosar, 2019-08-05:
I tried to implement the getDistanceSensorData(lidar_name, vehicle_name) method but I received the following error.

NameError: name 'DistanceSensorData' is not defined

Trace:
_Traceback (most recent call last):
  File "E:/Projects/D- Offset/Area Coverage/Coverage/AreaCoverage/AreaCoverage.py", line 399, in <module>
    handlerObj.main()
  File "E:/Projects/D- Offset/Area Coverage/Coverage/AreaCoverage/AreaCoverage.py", line 351, in main
    print(airSimObj.getDistanceSensorData(client=drone_client, vehicle_name=vehicles[0]))
  File "E:/Projects/D- Offset/Area Coverage/Coverage/AreaCoverage/AreaCoverage.py", line 291, in getDistanceSensorData
    client.getDistanceSensorData()
  File "C:\Program Files\Python36\lib\site-packages\airsim\client.py", line 157, in getDistanceSensorData
    return DistanceSensorData.from_msgpack(self.client.call('getDistanceSensorData', distance_sensor_name, vehicle_name))
NameError: name 'DistanceSensorData' is not defined_

I tried to find the DistanceSensorData class but couldn't find it. Has anyone faced a similar issue? And if so, how did you resolve it?
Any help is much appreciated

rajatthosar, 2019-08-09:
Is there any update on this issue?

ACLeighner, 2019-10-07:
I was able to fix this by adding the following code the the types file.

AirSim\PythonClient\airsim\types.py

class DistanceSensorData(MsgpackMixin):
    time_stamp = np.uint64(0)
    distance = Quaternionr()
    min_distance = `Quaternionr()`
    max_distance = Quaternionr()
    relative_pose = Pose()

saihv, 2019-10-30:
@ACLeighner please feel free to submit a PR!

PR #2403 (2020-03-23): Fix missing DistanceSensorData type and wrong param name in getDistanceSensorData()
* Wrong variable name in getDistanceSensor()
* Missing type for DistanceSensorData

Fixes #2117 

There is also a lack of documentation on how to use the distance sensors, would be nice if someone could fix that too. Already mentioned in #1673.
closed 2020-03-23 by madratman